veryl-lang/veryl #634: The translation from assign to always_comb resulted in an error

zrr1999, 2024-04-07:
I am a beginner, and I am using Verilog to write a simple RAM module:
```
module RAM (
    i_ram_ena  : input logic   ,
    i_ram_read : input logic   ,
    i_ram_write: input logic   ,
    i_ram_addr : input logic<8>,
    io_ram_data: inout logic<8>,
) {
    var memory: logic<8> [256];
    assign io_ram_data = if i_ram_read && i_ram_ena {
        memory[i_ram_addr]
    } else {
        8'hzz
    };

    always_ff (i_ram_write) {
        if i_ram_ena {
            memory[i_ram_addr] = io_ram_data;
        }
    }
}
```
This code was translated into:
```verilog
module RAM (
    input logic         i_ram_ena  ,
    input logic         i_ram_read ,
    input logic         i_ram_write,
    input logic [8-1:0] i_ram_addr ,
    inout logic [8-1:0] io_ram_data
);
    logic [8-1:0] memory      [0:256-1];
    always_comb io_ram_data = ((i_ram_read && i_ram_ena) ? (
        memory[i_ram_addr]
    ) : (
        8'hzz
    ));

    always_ff @ (posedge i_ram_write) begin
        if (i_ram_ena) begin
            memory[i_ram_addr] <= io_ram_data;
        end
    end
endmodule
```
When I use iverilog, I get the following error:
src/ram.sv:9: error: io_ram_data is not a valid l-value in RAM.
src/ram.sv:6:      : io_ram_data is declared here as wire.
However, when I change always_comb to assign, I don't get an error.
So, are there certain situations where assign should not be translated to always_comb?

dalance, 2024-04-07:
I missed this case.
`inout` can't be assigned through `always_comb`.
Additionally `inout logic` is invalid, `inout wire` should be emitted instead of it.

dalance, 2024-04-12:
I've added `tri` type modifier check for `inout` port,
and `assign` is emitted at assignment to `tri` type variable after merging #655.

```systemverilog
    io_ram_data: inout tri logic<8>, // add tri type modifier
```

PR #655 (2024-04-12): Add tri-state assignment support
Closes #634
closed 2024-04-12 by dalance