veryl-lang/veryl #641: [Question] How to use case statement with don't care?

nananapo, 2024-04-10:
Hi.
Is the case statement with don't care like ```casez``` or ```casex``` supported?
I want to use it to implement a decoder.

taichi-ishitani, 2024-04-10:
Veryl has not yet supported these statement.

@dalance 
SystemVerilog has `case inside` statement that is enhanced version of `casex` statement.
So I think Veryl should support `case inside` but has not to support `casex` and `casez` statements.

nananapo, 2024-04-11:
By the way, case expression seems to be converted into logic using ternary operator ```?```. But I think converting into case statement is better.

dalance, 2024-04-12:
> By the way, case expression seems to be converted into logic using ternary operator `?`. But I think converting into case statement is better.

The reason using ternary operator is that it can be used at arbitrary expression context like below:

```systemverilog
// pass
localparam A = X == 1 ? 0 : 1;

// fail
localparam A = case (X)
    1: 0;
    default: 1;
endcase
```

nananapo, 2024-04-12:
I think it would be nice if the semantics of case expression and statement are not different.

taichi-ishitani, 2024-04-12:
SystemVerilog does not have `if expression` and `case expression` features so I think it's difficult.

closed 2024-06-11 by dalance

taichi-ishitani, 2024-06-14:
@nananapo ,
I changed emmition of `case` statemen from the simple `case` statement to the `case inside` statement.
Now you can use unknown values for case conditions like the `casex` statement.

nananapo, 2024-06-14:
やったー😆
ありがとうございます🙏