veryl-lang/veryl #643: [Feature] `case inside` statement and expression

taichi-ishitani, 2024-04-11:
SystemVerilog has the `case inside` statement that is enhanced version of `casex` statetment.
Veryl should support it.

_Originally posted by @taichi-ishitani in https://github.com/veryl-lang/veryl/issues/641#issuecomment-2048010806_

dalance, 2024-04-11:
Emitting `case inside` instead of `case` from Veryl's `case` seems to be good?
Is there cases to require the original `case` emitting?

From

```systemverilog
case x {
    0      : a = 1,
    1..3   : a = 2,
    'h1xx  : a = 3,
    default: a = 4,
}
```

To

```systemverilog
case (x) inside {
    0      : a = 1,
    [1:2]  : a = 2,
    'h1xx  : a = 3,
    default: a = 4,
}
```

taichi-ishitani, 2024-04-11:
Like the exmaple below, behavior of `case` and `case inside` are different when a case item expression has unknwon value.

```systemverilog
case (a)
  b:      c = 1; // select only when a == b
  default c = 0;
endcase

case (a) inside
  b:      c = 1; // select when a == b or b is unknown
  default c = 0;
endcase
```

So I think these should be different syntax.

taichi-ishitani, 2024-04-11:
> Like the exmaple below, behavior of `case` and `case inside` are different when a case item expression has unknwon value.
>
> ```systemverilog
> case (a)
>   b:      c = 1; // select only when a == b [...]

This behavior may cause difference between simulation and synthesis result.
(refs: https://qiita.com/taichi-ishitani/items/d5bb34273fce8d4b0385#%E8%A4%87%E6%95%B0%E5%80%8B%E3%81%AE%E6%AF%94%E8%BC%83%E5%80%A4%E3%81%A8%E3%81%AE%E6%AF%94%E8%BC%83)

I think it can be solution for this problem to introduce limitation that `case item expr` should be a constant.

dalance, 2024-04-11:
I think the following `case` and `switch` seems to be good.

* `case` statement
  * case item should not be expression
  * emits `case inside`
* `switch` statement (refs #11)
  * switch arm can be expression
  * emits `case`

```systemverilog
case a {
    0      : c = 1;
    default: c = 2;
}

switch {
    a == b : c = 1;
    default: c = 2;
}
```

```systemverilog
case (a) inside
    0: c = 1;
    default: c = 2;
endcase

case (1)
    a == b : c = 1;
    default: c = 2;
endcase
```