veryl-lang/veryl #664: Wrong interface instantiation

nananapo, 2024-04-14:
```sv
interface ifA {
    var _a: logic;
}
module ModuleA {
    inst _x: ifA;
    var _y: ifA;
}
```
This Veryl code is translated into 
```sv
interface project_ifA;
    logic _a;
endinterface
module project_ModuleA;
    project_ifA _x ();
    project_ifA _y;
endmodule
```

Is ```var _y: ifA;``` legal in veryl?

If it's legal, 
According to IEEE Std 1800-2017 A.4.1.1 Module instantiation
```
hierarchical_instance ::= name_of_instance ( [ list_of_port_connections ] )
```
and A.4.1.2 Interface instantiation, 
```
interface_instantiation ::=
interface_identifier [ parameter_value_assignment ] hierarchical_instance { , hierarchical_instance } ;
```

I think interface instantiation can't omit ```()```.

taichi-ishitani, 2024-04-15:
> Is var _y: ifA; legal in veryl?

See #638

dalance, 2024-04-15:
It is not valid in Veryl.
Type of variable should be built-in types or user defined struct, enum, unoin, typedef, parameter type.
I'll add variable type check like below:

```
Error: mismatch_type (https://doc.veryl-lang.org/book/06_appendix/02_semantic_error.html#mismatch_type)

  × "ifA" is expected to "enum or union or struct", but it is "interface"
   ╭─[/home/hatta/work/repos/veryl/test.veryl:5:1]
 5 │     inst _x: ifA;
 6 │     var _y: ifA;
   ·         ─┬
   ·          ╰── Error location
 7 │ }
   ╰────
  help:
```

PR #666 (2024-04-15): Add variable type check
Closes #664
closed 2024-04-15 by dalance