veryl-lang/veryl #690: Support Dev Container Features

puppuccino, 2024-04-25:
Hi,
Have you thought about distributing the official Dev Container Features of Veryl? Veryl has features like real-time diagnostics and dependency management, which aren’t fully showcased in the playground. By distributing the Dev Container Features, anyone interested in Veryl could experience the full capabilities of this fantastic project with just a single click! The installation process is already straightforward, but it could be made even simpler.
As far as I have researched, there is a [template](https://github.com/devcontainers/feature-starter). You can modify it to publish the Dev Container Features.

dalance, 2024-04-25:
I created a repository from the template for now.

https://github.com/veryl-lang/devcontainer-features

dalance, 2024-04-26:
I published Veryl's feature:

 https://github.com/veryl-lang/devcontainer-features/pkgs/container/devcontainer-features%2Fveryl

It seems to work fine through GitHub Codespaces by editing `devcontainer.json`.
UI based selection will be enabled after merging https://github.com/devcontainers/devcontainers.github.io/pull/396.

dalance, 2024-04-30:
Veryl and Verilator feature are listed at the official index.

https://containers.dev/features