veryl-lang/veryl #708: [Feature] clock/reset port connection check

taichi-ishitani, 2024-05-09:
refs: https://github.com/veryl-lang/veryl/issues/622#issuecomment-2030848796

To prevent clock/reset type mismatch, we need to check type mismatch on clocl/reset port connections.

```systemverilog
module ModA (
  i_clk:   input clock_posedge,
  i_rst_n: input reset_async_low
) {}

module ModB (
  i_clk: input clock,
  i_rst: input reset
){
  inst u_moda: ModA (
    i_clk:   i_clk, // these connection including type mismatch
    i_rst_n: i_rst  // Veryl should report an error.
  );
}

nblei, 2024-06-04:
I'm a bit confused.

What does `clock_posedge` mean?  Does it mean that any flops driven by this clock must be positive edge-triggered?  In which case, what is the issue with connecting `i_clk : clock` to `i_clk : clock_posedge`?

nblei, 2024-06-04:
I get that, I think I'm more questioning the notion of this being a property of the _clock_ wire rather than a property of the registers.

taichi-ishitani, 2024-06-04:
I'd like to simplify `always_ff` declarations so I've moved clock/reset prorperties from `always_ff` block to `clock`/`reset` types.

dalance, 2024-06-05:
I think that using `clock_posedge` is more explicit than placing `posedge` in sensitivity list.
Espacially I like it can show that the clock/reset connection should be careful at the module port declaration which is public API.

```systemverilog
module ModuleA (
    i_clk: clock_posedge,
    i_rst: reset_async_low,
) {
    always_ff {
        if_reset {
        } else {
        }
    }
}
module ModuleB (
    i_clk: clock,
    i_rst: reset,
) {
    let w_clk: clock_negedge = i_clk;
    let w_rst: reset_async_high = i_rst;
    always_ff (w_clk, w_rst) {
        if_reset {
        } else {
        }
    }
}
```

```systemverilog
module ModuleA (
    i_clk: clock,
    i_rst: reset,
) {
    always_ff (posedge i_clk, async_low i_rst) {
        if_reset {
        } else {
        }
    }
}
module ModuleB (
    i_clk: clock,
    i_rst: reset,
) {
    always_ff (negedge i_clk, async_high, i_rst) {
        if_reset {
        } else {
        }
    }
}
```

PR #891 (2024-08-16): Check clock/reset port connection
Closes #708
closed 2024-08-16 by dalance