veryl-lang/veryl #711: Raw identifier

dalance, 2024-05-10:
Some Veryl's keyword can be used as identifier in SystemVerilog.
So instantiating a SystemVerilog module may cause syntax error.

```systemverilog
module ModuleA (
    i_clk: input clock,
    i_rst: input reset,
) {
    inst u0: $sv::ModuleSV (
        clock: i_clk, //syntax error
        reset: i_rst, //syntax error
    );
}
```

So Rust-like raw identifier may be useful.

```systemverilog
module ModuleA (
    i_clk: input clock,
    i_rst: input reset,
) {
    inst u0: $sv::ModuleSV (
        r#clock: i_clk,
        r#reset: i_rst,
    );
}
```

Raw identifier starting with `r#` will be transpiled to an identifier without `r#`.

nblei, 2024-06-28:
Dalance,

Rather than using `r#clock`, etc., is there a way to catch the parse error itself or have keywords conditioned by context?  The parser knows that it is in a `sv context`, then it can ignore Veryl keywords, perhaps.

dalance, 2024-06-28:
I think it is difficult to judge whether arbitrary identifier belongs sv context syntactically.
For example, the following case requires symbol table resolving to detect it.

```
inst u_if: $sv::SvInterface();

assign u_if.clock = 1;
```

Additionally, syntax highlighting becomes more difficult or impossible on some editors.

PR #806 (2024-07-01): Add raw identifier
Closes #711
closed 2024-07-01 by dalance