veryl-lang/veryl #739: Semantic error checker flags non-real errors

nblei, 2024-05-28:
The check for multiple assignments occurs at the variable level, rather than the bit/wire level.

```systemverilog
module fa (
    i_a : input  logic,
    i_b : input  logic,
    i_ci: input  logic,
    o_s : output logic,
    o_co: output logic,
) {
    assign o_s  = ^{i_a, i_b, i_ci};
    assign o_co = |{i_a & i_b, i_a & i_ci, i_b & i_ci};
}

module rca #(
    param WIDTH: u32 = 8,
) (
    i_a : input  logic<WIDTH>,
    i_b : input  logic<WIDTH>,
    i_ci: input  logic       ,
    o_s : output logic<WIDTH>,
    o_co: output logic       ,
) {
    var carries: logic [WIDTH + 1];

    assign carries[0] = i_ci;
    assign o_co       = carries[WIDTH];

    for i in 0..WIDTH :gen_fas {
        inst u_fa: fa (
            i_a : i_a[i]        ,
            i_b : i_b[i]        ,
            i_ci: carries[i]    ,
            o_co: carries[i + 1],
            o_s : o_s[i]        ,
        );
    }
}
```

```bash
  × carries is assigned in multiple procedural blocks or assignment statements
    ╭─[/home/nathan/Documents/Reviews/MICRO-2024/rca/src/rca.veryl:20:1]
 20 │ ) {
 21 │     var carries: logic [WIDTH + 1];
    ·         ───┬───
    ·            ╰── Error location
 22 │ 
 23 │     assign carries[0] = i_ci;
    ·     ───┬──
    ·        ╰── Assigned
 24 │     assign o_co       = carries[WIDTH];
 25 │ 
 26 │     for i in 0..WIDTH :gen_fas {
 27 │         inst u_fa: fa (
 28 │             i_a : i_a[i]        ,
 29 │             i_b : i_b[i]        ,
 30 │             i_ci: carries[i]    ,
 31 │             o_co: carries[i + 1],
    ·             ──┬─
    ·               ╰── Assigned too
 32 │             o_s : o_s[i]        ,
    ╰────
```

dalance, 2024-05-29:
Partial assignment at port connection seems to be treated as full assignment now.
I'll fix it.

closed 2024-05-29 by dalance

nblei, 2024-05-31:
Naoya,

This seems to have been a course correction too far!

```SystemVerilog
module rca #(
    param WIDTH: u32 = 8,
) (
    i_a : input  logic<WIDTH>,
    i_b : input  logic<WIDTH>,
    i_ci: input  logic       ,
    o_s : output logic<WIDTH>,
    o_co: output logic       ,
) {
    // carry-outs
    var carries: logic [WIDTH + 1];

    assign carries[0] = i_ci;
    assign carries[0] = ~i_ci;
    assign o_co       = carries[WIDTH];

    for i in 0..WIDTH :gen_fas {
        inst u_fa: full_adder (
            i_a : i_a[i]        ,
            i_b : i_b[i]        ,
            i_ci: carries[i]    ,
            o_co: carries[i + 1],
            o_s : o_s[i]        ,
        );
    }
}
```

Now this builds successfully, despite having multiple assignments to the same bit.  This is, I think, better than before, because the bug will get picked up during synthesis, but is still not ideal.

dalance, 2024-05-31:
Yes. The multiple assignment by partial assignment is not checked yet.
I created a new issue for it.

PR #740 (2024-05-29): Fix multiple assignment check for partial port connection
Closes #739