veryl-lang/veryl #754: Undefined identifier error for generic type argument

taichi-ishitani, 2024-06-05:
Veryl reports an undefined identifier error for a generic type argument pointing a type locally defined.
For exmaple:

```systemverilog
package PackageA {
  function FuncA::<T> -> T {
    var a: logic;
    a = 0;
    return a as T;
  }
}

module ModuleA {
  type my_logic = logic;
  let _a: logic = PackageA::FuncA::<my_logic>();
}
```

[image]

taichi-ishitani, 2024-06-10:
To get valid generated SV code, the original type declaration need to be copied to the generated package declaration.

dalance, 2024-08-16:
This is a known restriction. ref: https://doc.veryl-lang.org/book/05_language_reference/14_generics.html

> Additionally, the actual parameters should be accessible at the position of the generics declaration. For example, module names can be used as actual parameters because it is accessible through the whole project. On the other hand, local parameters can’t be used as actual parameters in many cases. This is caused by that the local parameters is not accessible from the potision of the generics declaration.

The above case should be written like below:

```systemverilog
package PackageA {
  type my_logic = logic;
  function FuncA::<T> -> T {
    var a: T;
    a = 0;
    return a;
  }
}

module ModuleA {
  let _a: logic = PackageA::FuncA::<PackageA::my_logic>();
}
```

Copying the type declaration will not work fine in many cases like below:

```systemverilog
module ModuleA #(
  param my_logic: type = logic,
) {
  let _a: logic = PackageA::FuncA::<my_logic>();
}

module ModuleA #(
  param TYPE: type = logic,
) {
  type my_logic = TYPE;
  let _a: logic = PackageA::FuncA::<my_logic>();
}

```

dalance, 2024-10-13:
I'll close this issue because there is no way to handle the situation.
If you have another idea, please reopen.

closed 2024-10-13 by dalance