veryl-lang/veryl #793: [Feature] Generics for code block

taichi-ishitani, 2024-06-20:
SV's text macro is useful feature to reduce coding cost for repeated code blocks.
I need this kind of feature to import code below to Very.
https://github.com/rggen/rggen-sample/blob/b0439abf7d5fa6086c4d0aef0ffcdc6836506f06/uart_csr.sv#L1-L9

I think it can be achieved if the generics feature handle code blocks.

dalance, 2024-06-20:
A syntax idea about generic `block` declaration.

```systemverilog
block rggen_connect_bit_field_if::<RIF, FIF, LSB, WIDTH> {
    assign  FIF.valid                 = RIF.valid;
    assign  FIF.read_mask             = RIF.read_mask[LSB+:WIDTH];
    assign  FIF.write_mask            = RIF.write_mask[LSB+:WIDTH];
    assign  FIF.write_data            = RIF.write_data[LSB+:WIDTH];
    assign  RIF.read_data[LSB+:WIDTH] = FIF.read_data;
    assign  RIF.value[LSB+:WIDTH]     = FIF.value;
}

module ModuleA {
    inst u_block: rggen_connect_bit_field_if::<bit_field_if, bit_field_sub_if, 0, 8>;
}
```

taichi-ishitani, 2024-06-20:
It looks good.
But why is instance name need? Is it to separate namespace?

dalance, 2024-06-20:
> It looks good. But why is instance name need? Is it to separate namespace?

Yes. And adding name-less instantiation increases syntax complexity.
The generated SV will become like below:

```systemverilog
if (1) begin: u_block
    assign  FIF.valid                 = RIF.valid;
    assign  FIF.read_mask             = RIF.read_mask[LSB+:WIDTH];
    assign  FIF.write_mask            = RIF.write_mask[LSB+:WIDTH];
    assign  FIF.write_data            = RIF.write_data[LSB+:WIDTH];
    assign  RIF.read_data[LSB+:WIDTH] = FIF.read_data;
    assign  RIF.value[LSB+:WIDTH]     = FIF.value;
end
```

taichi-ishitani, 2025-01-29:
Module instance and code block instance are completely different.
I think another syntax should be introduced for code block instnace.

taichi-ishitani, 2025-01-30:
How about `block` or `insert`?

```systemverilog
block u_block: rggen_connect_bit_field_if::<bit_field_if, bit_field_sub_if, 0, 8>;
insert u_block: rggen_connect_bit_field_if::<bit_field_if, bit_field_sub_if, 0, 8>;
```

dalance, 2025-01-30:
In your example, almost all syntax except keyword seems to be the same as instance declaration.
And instantiation is not specialized for module, but also it is used for interface instantiation.
So I feel expanding it for block may be natural.

Could you explain about necessity of specialized syntax?

taichi-ishitani, 2025-01-30:
Main reason is a implementation issue.

Code block should be analyzed on the context where it is instantiated.
For exmaple:
https://github.com/veryl-lang/veryl/blob/e69e0051176212d412e70f7245800044076d6974/crates/analyzer/src/handlers/check_type.rs#L341

To do this, the implementation may be below.

```rust
fn inst_declaration(&mut self, arg: &InstDeclaration) -> Result<(), ParolError> {
  match symbo.found.kind {
    CodeBlock => {
      self.statement_block_group(&x.statement_block_group);
    }
    _ => {
      // copy original implemnetation here
    }
  }
}
```

We need to insert the above code to analyzer handlers:

* Execute context aware check for code block
* Override `inst_declaration` method

I think this is troublesome so I'd like to separate `InstDeclaration` rule for code block and module instance.