veryl-lang/veryl #803: duplicate reserved [clock] word in vendor IP core.

YSawc, 2024-06-28:
I use vendor IP core for developing memory system, but reserved `clock` word is duplicated with veryl.
Is there already some way to avoid duplication in this case ? 
```
    inst ip_wram: $sv::ip_wram (
        address: i_addr,
        clock  : i_casted_clk, 
        data   : i_data        ,
        wren   : i_ip_wram_we  ,
        q      : o_ip_wram_data,
    );
```

compile error here.
```
Error: ParserError::SyntaxError

  × Unexpected token: ','
    ╭─ ...
 14 │     inst ip_wram: $sv::ip_wram (
 15 │         address: i_addr,
    ·                        ┬
    ·                        ╰── Error location
 16 │         clock  : i_casted_clk,
    ╰────
  help:

```

dalance, 2024-06-28:
This is planed at #711.
I'll implement it in the next week.

dalance, 2024-07-01:
I merged #806.

closed 2024-07-01 by YSawc