veryl-lang/veryl #811: inside of case is not implemented quartus

YSawc, 2024-07-04:
I use latest quartus prime lite for program logic but the software is not supported inside of case yet.

generated.sv
```
       case (status) inside
            Status_idle: ...
        endcase
```

quartus compile error here.
```
Error (10170):... near text: "inside";  expecting an operand. Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number.

```

Like this case is not offten but, I think better if there is something like compile flag controlled by veryl.toml example.

dalance, 2024-07-05:
Veryl's case supports range and don't care, so removing `inside` from generated SV doesn't work fine.
Instead of it, `if ( .. inside ..)` chain may be useful.

```
// default
case (status) inside
    Status_idle: ...
endcase

// "emit_if_inside_for_case = true" in Veryl.toml
if (status inside {Status_idle}) begin
end else if ... 
```

Is such usage of `inside` supported by Quartus?

taichi-ishitani, 2024-07-05:
The `inside` operator is also unsupported.
https://www.intel.com/content/www/us/en/programmable/quartushelp/current/index.htm#hdl/vlog/vlog_list_sys_vlog_d1627e719.htm

YSawc, 2024-07-05:
As a taichi-shitani said `inside` operators not supported.

generated.sv
```
    Status         status;
    typedef enum logic  [3-1:0] {
        Status_Ready,
        Status_ReadTargetLowAddr,
        Status_ReadTargetHighAddr,
...
        Status_Finish
    } Status;
...
            if (status inside {Status_Ready} ) begin
                ...
            end else begin
                ...
            end
```

build error message here.
```
Error (10170): ... near text: "inside";  expecting ")". Check for and fix any syntax errors that appear immediately before or at the specified keyword. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number.
```

Inside expression is not supported yet in quartus, so instead of `inside`, conditional operation works fine.
```
if (status >= Status_Ready && status <= Status_ReadTargetHighAddr ) begin
  if_register.a <= 0;
end else begin
  if_register.a <= 1;
end
```

In case of using inside, can it cover with like above range expression instead of `inside` ?
Or if another better solution exists, please tell me, thank.

taichi-ishitani, 2024-07-05:
refs: #145

taichi-ishitani, 2024-07-05:
How about emitting `case` statement instead of `case inside` statement if all case condition expressions do not include range expression and unknown values?

YSawc, 2024-07-06:
It sounds good.

nathsou, 2024-07-24:
Hi, running a cocotb testbench with iverilog on a module with a `case inside` also fails.

taichi-ishitani, 2024-08-13:
I think this issue is fixed by #873.

closed 2024-08-15 by dalance