veryl-lang/veryl #817: clock_domain for always_ff

prokie, 2024-07-15:
I noticed the clock domain annotation which I like, but it does not work for `always_ff`.

Is that something that will be supported in the future?

```veryl
module ModuleA (
    // belong clock domain 'a
    i_clk_a: input  'a clock,
    i_dat_a: input  'a logic,
    o_dat_a: output 'a logic,

    // belong clock domain 'b
    i_clk_b: input  'b clock,
    i_dat_b: input  'b logic,
    o_dat_b: output 'b logic,
) {

    always_ff {
        o_dat_a = i_dat_a;
    }

    always_ff {
        o_dat_b = i_dat_b;
    }
}
```

dalance, 2024-07-16:
Omitting clock of `always_ff` can be used only when there is single clock in the module.
In this case, there are two clocks, so clock specification is required like below.

```systemverilog
module ModuleA (
    // belong clock domain 'a
    i_clk_a: input  'a clock,
    i_dat_a: input  'a logic,
    o_dat_a: output 'a logic,

    // belong clock domain 'b
    i_clk_b: input  'b clock,
    i_dat_b: input  'b logic,
    o_dat_b: output 'b logic,
) {

    always_ff (i_clk_a) {
        o_dat_a = i_dat_a;
    }

    always_ff (i_clk_b) {
        o_dat_b = i_dat_b;
    }
}
```

prokie, 2024-07-16:
Aa I had misunderstood the purpose of the clock domain annotation, but now I understand. Thanks :)

closed 2024-07-16 by prokie