veryl-lang/veryl #823: Register initialization

nathsou, 2024-07-22:
Hi, thanks for the good work!

Maybe I missed something, but I couldn't find an example of how to initialize a register value:

```veryl
pub module RegisterFile (
  i_clk: input clock,
) {
  var regs: logic<16>[8];

  initial {
    regs[0] = '0;
  }
}
```

Fails with:

```
Error: invalid_statement (link)

  × assignment statement can't be placed at here
   ╭─[/home/nathan/code/fpga/cpu16/design/src/RegisterFile.veryl:7:13]
 6 │   initial {
 7 │     regs[0] = '0;
   ·             ┬
   ·             ╰── Error location
 8 │   }
   ╰────
  help: remove assignment statement
  ```
  
  Is there a way to achieve this already?
  
  Thanks

nblei, 2024-07-23:
You initialize registers via reset.

```systemverilog
always_ff {
  if_reset { 
    regs[0] = '0;
  } else {
   ...  
  }
}
```

nathsou, 2024-07-23:
Thanks @nblei for your quick reply, while this indeed works, using a reset signal does not seem behaviorally equivalent as we need to introduce an input reset signal in the module ports and assert it high at startup.

Is this a desired design decision? I guess this is not a bad thing since it means that the design can easily be ported to devices without initialization capabilities like ASICs.

Although initialization with `$readmemb` and `$readmemh` seems to work fine, so this does not seem to be the case:

```Veryl
  var regs: logic<16>[8];

  initial {
    $readmemh("regs.hex", r_regs);
  }
  ```

nblei, 2024-07-23:
Veryl is a hardware description language --- not a simulation language.   As flops take random values on power-on, initializing them in an `initial` procedural block does not make sense.

$readmemh "works" because SystemVerilog system functions get transpiled literally.

nathsou, 2024-07-23:
I'm not at all an expert but initial blocks are not just used for simulation, some FPGAs do support flip-flop initialization, I have for instance used `$readmemh` to initialize a block RAM with a simple image and when actually programmed onto a Xilinx 7 FPGA in this case, the image did appear so the flip flops were in the correct startup state without relying on a reset signal.

Anyway the proposed solution does work although it's a workaround, so I'm closing this issue.

nblei, 2024-07-23:
>I have for instance used $readmemh to initialize a block RAM with a simple image and when actually programmed onto a Xilinx 7 FPGA in this case

FPGA BRAM is implemented as SRAMs, not flops.

nathsou, 2024-07-23:
Makes sense

punzik, 2024-11-09:
> Veryl is a hardware description language --- not a simulation language. As flops take random values on power-on, initializing them in an `initial` procedural block does not make sense.
 
In most FPGAs, the initial state of the triggers is determined by the configuration bitstream. Therefore, the initial initialization is successfully synthesized for the FPGA. This is convenient when, for example, there is no external reset signal. 
I think it would be right to add the possibility of initial initialization in Veryl.

dalance, 2024-11-11:
I agree that this use case is common in FPGA. But I think allowing this feature unconditionally is not good. This is because ASIC synthesizers (possibly few FPGA synthesizer too?) ignore `initial` statement.
Now `readmemh` is allowed, but it is because assignment by function call is not checked yet. So I think it should be checked in future, and the way which allows this feature explicitly should be provided for FPGA usage.

punzik, 2024-11-11:
@dalance, thanks for the reply. One solution is to create a Verilog module with a register and necessary initialization. It is only need to tell the module about clock polarity and reset type. Is it possible as I know?

dalance, 2024-11-12:
By following code, clock polarity and reset type which Verilog module receives can be specified.
Veryl compiler inserts `~` if polarity inversion is required at casting.

```systemverilog
module Module (
    i_clk: input clock,
    i_rst: input reset,
) {
    let w_clk: `_ clock_posedge    = i_clk as clock_posedge;
    let w_rst:    reset_async_high = i_rst as reset_async_high;

    inst u: $sv::ModuleSv (
        i_clk: w_clk,
        i_rst: w_rst,
    );
}
```

taichi-ishitani, 2025-03-02:
How about emitting `if_reset` statement into `initial` statement?

Veryl

```systemverilog
always_ff {
  if_reset { 
    regs[0] = '0;
  } else {
    // do something
  }
}
```

SV

```systemverilog
initial begin
  regs[0] = '0;
end

always @(posedge i_clk) begin
  // do something
end
```

dalance, 2025-03-04:
I imagine FPGA users want to use `readmemh` to initialize ROM.
So keeping to separate `initial` and `if_reset` may be better.

taichi-ishitani, 2025-03-27:
> I imagine FPGA users want to use `readmemh` to initialize ROM. So keeping to separate `initial` and `if_reset` may be better.

I think initializing an array from file input is useful for ASIC development.
How about adding functions similar to `$readmemh` and `$readmemb` functions and emitting an array literal from the file input given to the such functions?

Veryl

```systemverilog
always_ff {
  if_reset {
    regs = readmemh("fw.hex");
  } else {
    ...
  }
}
```

For ASIC

```systemverilog
always_ff @(posedge i_clk, negedge i_rst_n) begin
  if (!i_rst_n) begin
    regs = '{'h00, 'h01, ...};
  end else begin

  end
end
```

For FPGA

```systemverilog
initial begin
  $readmemh(regs, "fw.hex");
end

always @(posedge i_clk) begin
  ...
end
```

dalance, 2025-04-18:
I think embedding the contents to SV is not good.
This is because "fw.hex" can be changed after Veryl compilation.
It causes mismatch between ASIC and FPGA, and it is difficult to find it.

nblei, 2025-04-24:
These 
```
always_ff @(posedge i_clk, negedge i_rst_n) begin
  if (!i_rst_n) begin
    regs = '{'h00, 'h01, ...};
  end else begin

  end
end
```
initial begin
  $readmemh(regs, "fw.hex");
end

always @(posedge i_clk) begin
  ...
end
```
have very different semantics.

ROMs in ASICs can be implemented in several ways, including mask ROMs, EEPROMs, and in random logic.  The only ROM whose data would ever be described in synthesizable RTL is a logic ROM.  For these, there could be a new functionality added, something like:

```SystemVerilog
 #[generate_rom(LittleEndianOctet)]
let rom: logic<32> [256] = VerylReadMem(file.hex);
```

which would translate to
```SystemVerilog
logic [31:0] rom [256];
assign rom[0] = <first word in file.hex>;
assign rom[1] = <second word file.hex>;
// etc
```

This would be in line with Veryl's generics to support RTL generation.

dalance, 2026-09-01:
#3223 provides the way to initialize register from FPGA bitstream.

closed 2026-09-01 by dalance