veryl-lang/veryl #861: [Feature] Rewrite `inside` expression and `case` statement

taichi-ishitani, 2024-08-08:
There are tools not supporting `inside` expression and `case inside` statement.
(refs #811)

For these tools, Veryl should have an option to rewrite `inside` expression and `case inside` statement.

taichi-ishitani, 2024-08-08:
It seems that Quartus and iverilog support the wildcard operator `==?`.
https://www.intel.com/content/www/us/en/programmable/quartushelp/17.0/mapIdTopics/jka1465580570693.htm
https://github.com/steveicarus/iverilog/blob/676b36e4554f119d0606ade79db2fba6834d3445/ivtest/ivltests/wild_cmp_net.v#L6

taichi-ishitani, 2024-08-08:
This `case` statement can be emitted like below.

```
    always_comb {
        case x {
            0: a = 1;
            1: a = 1;
            2: {
                   a = 1;
                   a = 1;
                   a = 1;
               }
            3, 4   : a = 1;
            5..=7  : a = 1;
            y - 1  : a = 1;
            default: a = 1;
        }
    }
```

```systemverilog
always_comb begin
  case (1'b1)
    x ==? 0: a = 1;
    x ==? 1: a = 1;
    x ==? 2: begin
      a = 1;
      a = 1;
      a = 1;
    end
    a ==? 3, a ==? 4:     a = 1;
    (a >= 5) && (a <= 7): a = 1;
    a ==? (y - 1):        a = 1;
    default:              a = 1;
   
  endcase
end
```

taichi-ishitani, 2024-08-08:
This `inside` oepration can be emitted:

```
assign a = inside 1 + 2 / 3 {0, 0..10, 1..=10};
```

```systemverilog
assign a = ((1 + 2 / 3) ==? 0) ||
           ((1 + 2 / 3) >= 0 && (1 + 2 / 3) < 10) ||
           ((1 + 2 / 3) >= 0 && (1 + 2 / 3) <= 10);
```

dalance, 2024-08-08:
Looks good.
I think both expansion is acceptable.

closed 2024-08-13 by dalance