veryl-lang/veryl #919: `let` declaration in case item witout block

taichi-ishitani, 2024-08-30:
Currently, a `let` declaration in a case item without block like below can be parsed and veryl report a `latch generation` error.

```
// module definition
module ModuleA (
    i_sel : input  logic            
) {
  always_comb {
    case i_sel {
      0: let _a:logic = i_sel;
      1: let _b:logic = i_sel;
    }
  }
}
```

[image]

This kind of code is meaningless and the error message is confusable.
Therefore, I think Veryl should report a parse error for this kind of code.

nblei, 2024-09-01:
This also occurs in if statements.

taichi-ishitani, 2024-09-01:
[image]

Yes, this error message is unclear.
So Veryl should report error for a statement block without statements.

nblei, 2024-09-01:
Ignoring the scoping aspect of this for a moment, this issue stems from a lack of symbolic execution or other code coverage mechanism.   If Veryl does not have a mechanism by which to symbolically evaluate conditionals, how can it know that a latch is not inferred?  Currently, it is just checking syntactically --- `d` is assigned in the `if` block, but not in the `else` block or in an outer scope.

So it seems that there are a number of options:

1. remove latch checks.
2. change the warning to something lighter like "it may cause latch generation".
3. implement some form of coverage checking.

My vote would be for '2' now and sometime in the future, '3'.

The scoping question is also interesting, though.  Can you even define a variable in an inner scope in a SystemVerilog procedural block?

taichi-ishitani, 2024-09-01:
After merging #921, Veryl will not report the latch generation error for a variable declared in a statement block. See the unit test code below.

https://github.com/veryl-lang/veryl/blob/d1177bbcd10eabc31843784cf4b3fd199c3e1f54/crates/analyzer/src/tests.rs#L1790-L1818

taichi-ishitani, 2024-09-02:
Should Veryl emit a variable declaration within a statement block as an `automatic` variable?

closed 2024-09-02 by dalance