veryl-lang/veryl #926: Empty statement block

taichi-ishitani, 2024-09-02:
Currently, Veryl reports the `unused variable` error for a statement block including var declarations but no statements (refs: https://github.com/veryl-lang/veryl/issues/919#issuecomment-2323346711). In addition, Veryl reports no errors for an empty statement block.

Should Veryl report the `empty statement block` error for these cases?

dalance, 2024-09-03:
I don't have any idea about a special reason of reporting empty statement block as error.
How do you think?

taichi-ishitani, 2024-09-03:
I think the `unused_variable` error for the statement block including variable declarations only is enough.
This is the current implementation.

@nblei ,
What do you think about this topic?

nblei, 2024-09-03:
@taichi-ishitani is there any practical use case for an empty statement block?

dalance, 2024-09-03:
I think there is no intentional use case, but it temporally occur by comment-out during implementing or debugging.

```systemverilog
always_ff {
    if a {
        //x = 0;
    } else if b {
        x = 1;
    }
}
```

So I think warning is better than error.

taichi-ishitani, 2024-09-03:
I use an empty block with case statement.

```systemverilog
case (foo)
  A: do_a_action();
  B: do_b_action();
  C: ; // nothing to do
  default: do_default_action();
endcase
```

If am empty block is forbidden I need to re-write above code like below.

```systemverilog
case (foo)
  A: do_a_action();
  B: do_b_action();
  default: begin
    if (foo != C) begin
      do_default_action();
    end
  end
endcase
```

There is a case that an empty block can help to reduce block nest depth.

closed 2024-09-30 by dalance