veryl-lang/veryl #931: Check combinational loop

taichi-ishitani, 2024-09-03:
Combinational loop should be resolved but Veryl does not check it.

[image]

nblei, 2024-11-01:
How general should combinational loop checking be?

Should veryl compiler fully elaborate and flatten the design, including all generate blocks, to identify combinational loops?

How do inout ports get handled?

This is a very big task, I think.  But also a useful one.  It would be very cool to see combinational loop detection via language server.

taichi-ishitani, 2024-11-01:
> Should veryl compiler fully elaborate and flatten the design, including all generate blocks, to identify combinational loops?

Veryl compiler is a just transpiler so it's difficult to check combinational loops including paths crossing module boundaries as you mention.
I'm implementing feature to track all of assignments and references in PR #1036. I think we can trace combinational paths not crossing a module boundary by using this feature.

nblei, 2024-11-03:
Within module would still require complexities for generate blocks.

For example,

```systemverilog
assign v[0] = v[31];
for i in 1..32 {
  assign v[i] = v[i-1];
}
```
contains a combinational loop.

dalance, 2025-11-22:
Probably, this feature can be realized after #1847, because all instantiation, generate, function are expanded in IR construction phase.
#1287 will generate DAG of variables to decide evaluation order, so the DAG can detect combinational loop.