veryl-lang/veryl #937: Add `unique`, `unique0`, and `priority` keywords

nblei, 2024-09-04:
Consider
```systemverilog
module test (
	input [2:0] onehot,
	output logic [1:0] idx
);
always_comb begin
	case (onehot)
		3'b001: idx = 2'b01;
		3'b010: idx = 2'b10;
		3'b100: idx = 2'b11;
	endcase
end
endmodule : test
```
Synthesizing this with `dc_shell` results in inferred latches (ignore for a moment the fact that DC doesn't error out when latches are inferred in an `always_comb` block) because only three out of eight possible mux selects are handled in the switch statement.
This can be fixed by adding a default case, or by adding a default assignment prior to the switch statement.  However, this infers unnecessary logic when we are guaranteed the external controllability constraint that `onehot` is, in fact, one-hot.

However, adding the `unique` keyword solves this problem optimally:
```systemverilog
	unique case (onehot)
		3'b001: idx = 2'b01;
		3'b010: idx = 2'b10;
		3'b100: idx = 2'b11;
	endcase
```
The `unique case` is a mechanism for the RTL designer to encode controllability constraints.  It encodes the designer's guarantee to the synthesis tool / simulator / model checker that for each evaluation of the switch control signal, exactly one case in the switch statement will match.  Similarly, `unique` can be used to qualify `if-else if*-else` statements.

Other SV2009 keywords `unique0` and `priority` also modify the semantics of control statements.

taichi-ishitani, 2024-09-04:
`Switch` statement can also have these modifiers.

----

A synthesis tool can make very aggressive optimization for a case/if statement with `unique` modifier and this optimization may cause simulation/synthesis result mismatch when multiple conditions are matches.To prevent mismatch, our company has a lint rule to prohibit using `unique` modifier.
Thefore, I think we need to add a build option to prohibit these modifiers.

dalance, 2024-09-05:
At the point of view of syntax complexity, adding `#[cond_type]` attribute instead of adding keywords may be better.

```systemverilog
#[cond_type(unique)]
case a {
}

#[cond_type(priority)]
case a {
}

#[cond_type(unique0)]
if a {
} else if b {
} else {
}
```

About condition overlapping, mismatch between simulation and synthsis can be avoided if these modifiers are allowed when all conditions can be checked by Veryl compiler statilcally.

```systemverilog
param X: u32 = 1;
const Y: u32 = 1;

// OK
#[cond_type(unique0)]
case a {
    0: x = 1;
    Y: x = 2;
}

// NG because X may be overridden
#[cond_type(unique0)]
case a {
    X: x = 1;
}
```

taichi-ishitani, 2024-09-05:
> About condition overlapping, mismatch between simulation and synthsis can be avoided if these modifiers are allowed when all conditions can be checked by Veryl compiler statilcally.

I think Veryl also needs to check that all possible values for the case condition expression are listed as case items.

nblei, 2024-09-05:
There should not be a mismatch between synthesis and simulation in the sense that simulation will emit an error (or perhaps warning, I forget) in the case that no cases match, or more than a single case matches.

I do like the idea of a compiler switch which automatically removes the non-priority `cond_type` qualifiers.

I do not see how Veryl can possibly be expected to check condition overlapping.

Consider a microprocessor design which purposely does not check for illegal instructions (e.g., a microcontroller executing firmware embedded in an SoC).  A mux may only need to handle valid opcodes, but how is Veryl to know this?  In the presence of _external_ controllability don't cares, I think static analysis in Veryl is insufficient.

dalance, 2024-09-09:
> I do not see how Veryl can possibly be expected to check condition overlapping.

Yes. Veryl can't check overlapping in all cases.
So at the most safe side, there is an idea that these attribute can be used only when Veryl can check overlapping.
For example, if all case condition are numeric literal or const value refering literal, Veryl can check it.
I think these limitation may be acceptable.

nblei, 2024-09-09:
> So at the most safe side, there is an idea that these attribute can be used only when Veryl can check overlapping.

I think this is exactly the wrong approach.

If Veryl can check these conditions, then any logic synthesis tool will, also.

These modifiers exist explicitly to enable encoding of external controlability don't cares.  That is, they exist to be used precisely when logic synthesis tools (and thus Veryl) cannot check condition overlapping, condition uniqueness, etc.

dalance, 2024-09-10:
I understood like below. Is this right?

* When condition is static and overlapping check can be done by tools, synthesizer can optimize enough without these keywords.
* When these checking can't be done, designer notify the condition uniqueness to synthesizer through these keywords. This enables additional optimization which is impossible without these keywords.
* If there is any overlapping contrary to the keyword, mismatch between simulation and synthesis occurs. Designer is responsible for the condition ensurance.
* If these keywords are removed by compiler switch, mismatch doesn't occur, but some optimizations may be disabled.

nblei, 2024-09-10:
Yes on points 1, 2, 4.

Point 3, I disagree with the statement that there is mismatch between
simulation and synthesis.

In simulation, if `unique` is used and zero or more than 1 case match (or,
if zero or more than one if condition evaluates to true), then an
error/warning is emitted.
In simulation, if `unique0` is used and more than 1 case match (or, if more
than one if condition evaluates to true), then an error/warning is emitted.

dalance, 2024-09-10:
I agree wrong usage of these keywords emits warning, but in many cases simulator works fine with warning messages.
(Of course, we can configure to promote some warnings to hard errors, but it is not default behavior.)
So if we miss these messages, I think mismatch occurs.

taichi-ishitani, 2024-09-10:
> I think we need to add a build option to remove these modifires during emitting SV RTL

Regarding this, we cannot remove these quarifiers automatically if no default item is specified.
If we remove `unique` keyword from nblei's exmaple below,

```systemverilog
always_comb begin
	case (onehot)
		3'b001: idx = 2'b01;
		3'b010: idx = 2'b10;
		3'b100: idx = 2'b11;
	endcase
end
```

synthesis tools will infer latch logic.
Therefore, I think we need to tell Veryl which case item is default item to make possible to remove qualifiers automatically and safely.

```systemverilog
always_comb {  
  #[cond_type(unique)]
  case onehot {
    3'b001: idx = 1;
    3'b010: idx = 2;
    #[default]
    3'b100: idx = 3;
  }
}
```

taichi-ishitani, 2024-09-10:
We have to trust designer if RTL code including `unique case` works correctly.
Therefore, I think Veryl should report information message when `unique` quialifier is used.

dalance, 2024-09-12:
> We have to trust designer if RTL code including case/if statements with these quialifiers works correctly.
> Therefore, I think Veryl should report information message when quialifiers are used.

I think clean RTL should have no message because messages shown at all time will be ignored.
So how about the following designs?

* By default, `case_type` is ignored. If `emit_case_type = true` in `Veryl.toml`, it is emitted.
`emit_case_type = true` means that we understand risk of `case_type`, so the additional message is not required.

* By default, `case_type` is emitted and the additional message is shown.
After confirming safety of `case_type`, we set `suppress_case_type_message = true` in `Veryl.toml`, and the messages are suppressed.

taichi-ishitani, 2024-09-12:
I agree with this idea.
By the way, should Veryl support adding these quialifier to `if` statement?

nblei, 2024-09-12:
> By the way, should Veryl support adding these quialifier to if statement?

Yes.

taichi-ishitani, 2024-09-12:
> > By the way, should Veryl support adding these quialifier to if statement?
> 
> Yes.

For if statement, how should we add `default` attribute?

nblei, 2024-09-12:
The trailing `else` branch can be the default automatically.

taichi-ishitani, 2024-09-12:
How about if statement without else branch like this?

```systemverilog
if onehot[0] {
  idx = 1;
} else if onehot[1] {
  idx = 2;
} else if onethot[2] {
  idx = 3;
}
```

nblei, 2024-09-12:
With unique it is semantically equivalent to the version where the last
block is unconditional.

Without unique, we can just default the last block to be unconditional.

taichi-ishitani, 2024-09-13:
OK, I agree.
I think the following SVA should be inserted to the default item for ease to report the designer that no case conditions are matched.

```systemverilog
case (onehot)
  3'b001: idx = 1;
  3'b010: idx = 2;
  default: begin
    assume (onhot == 3'b100)
    else $error("unexpected condtion is matched");
    idx = 3;
  end
endcase

if (onehot[0]) begin
  idx = 1;
end else if (onehot[1]) begin
  idx = 2;
end else begin
  assume (onhot == 3'b100)
  else $error("unexpected condtion is matched");
  idx = 3;
end
```

nblei, 2024-09-13:
That works with the caveat that it does not produce error on uniqueness violation.  So if more than one case or branch evaluates to true, no error or warning is emitted.  Perhaps this is fine, though.

dalance, 2024-10-26:
I'm trying to implement this feature. The current spec is below:

* Add attributes: 
  * `#[cond_type(unique)]`
  * `#[cond_type(unique0)]`
  * `#[cond_type(priority)]`
  * `#[cond_type(none)]` 

`none` is used to control nested case like below:

```systemverilog
always_comb {  
  #[cond_type(unique)]
  case a {
    0: {
      #[cond_type(none)] // if there is not this attribute, "case b" is unique too.
      case b {
      }
    }
  }
}
```

* Add `emit_cond_type` config to `Veryl.toml`
If `emit_cond_type = true` is specified explicitly, `unique`, `unique0` and `priority` are emitted.
If not, they are not emitted, and the last case/if item becomes `default` to prevent unexpected latch generation.
`#[default]` attribute is not adopted because it requires reorder items, and it is difficult to add to if statement.

PR #1043 (2024-11-13): `cond_type` attribute
Closes #937
closed 2024-11-13 by dalance