veryl-lang/veryl #947: [Feature] `always_*` blocks inside interface

taichi-ishitani, 2024-09-06:
SV interface can have `always_*` blocks.
Veryl should also support this feature.

dalance, 2024-09-08:
I have no experience of these usage. Could you explain actual use case?

taichi-ishitani, 2024-09-08:
An `ack` logic is commnly used for a handshake protocol.
By supporting this feature, this `ack` logic can be declared in an interface like below.

```systemverilog
interface foo_if;
  logic       ready;
  logic       valid;
  logic [7:0] data;
  logic       ack;

  always_comb ack = ready && valid;

  modport master (
    input  ready,
    output valid,
    output data,
    input  ack
  );

  modport slave (
    output ready,
    input  valid,
    input  data,
    input  ack
  );
endinterface
```

And it can be used like below.

```systemverilog
alwyas_ff @(posedge i_clk) begin
  if (foo_if.ack) begin
    data <= foo_if.data;
  end
end
```

I think this feature can help to increase readability and maintainability.

dalance, 2024-09-09:
Thanks!
I understood the use case.

PR #950 (2024-09-09): Merge `ModuleItem` and `InterfaceItem`
For #793, generate constructions for module and interface need to be merged.
To do this, `GenerateItem` will be introduced and all items except for `ModportDeclaration` will be moved from `ModuleItem` and `InterfaceItem` to `GenerateItem`.

By merging `ModuleItem` and `InterfaceItem`, interface will be able to have interface instances (#908) and `always_*` blocks (#947).

resolve #947
closed 2024-09-09 by dalance