veryl-lang/veryl #952: Can't resolve type defined in package

nananapo, 2024-09-08:
eei.veryl
```sv
package eei {
    type Addr   = logic<32>;
}
```

membus_if.veryl
```sv
import eei::*;

interface membus_if {
    var addr  : Addr    ;
    // var addr  : eei::Addr    ; // <- OK
    modport master {
        addr  : output,
    }
}
```

core.veryl
```sv
import eei::*;

module core (
    ibus: modport membus_if::master,
) {
    let if_pc          : Addr  = 0;
    always_comb {
        ibus.addr  = if_pc;
    }
}
```

```
Error:   × veryl check failed

Error: undefined_identifier (link)

  × Addr is undefined
   ╭─[/home/kanataso/Documents/bluecore/sample/core.veryl:8:9]
 7 │     always_comb {
 8 │         ibus.addr  = if_pc;
   ·         ────┬────
   ·             ╰── Error location
 9 │     }
   ╰────
  help: 
```

dalance, 2024-09-10:
This is because import is interpreted in `create_reference` handler.

* Pass1: create_symbol_table
* Pass2: create_reference (including import interpretation)

So if `core.veryl` is analyzed before `membus_if.veryl`, resolving `ibus.addr` reaches `Addr` in `membus_if.veryl`, but the `Addr` is not imported because import in `membus_if.veryl` is not interpreted yet, and the `Addr` causes `undefined_identifier` error.

To resolve this issue, import interpretation should be done before create_reference like below:

* Pass1: create_symbol_table
* Pass2: create_import
* Pass3: create_reference

dalance, 2024-09-11:
Adding new pass only for import handling is too heavy.
Instead of it, adding post_pass1 for import is better.

* Pass1: create_symbol_table with gathering import information
* PostPass1: apply gathered import information
* Pass2: create_reference (can refer complete import information)

PR #957 (2024-09-11): Add post_pass1 stage for import handling
Closes #952
closed 2024-09-11 by dalance