veryl-lang/veryl #963: Interface prototype

dalance, 2024-09-13:


dalance, 2024-09-20:
Interface prototype may be the same as concrete interface in almost cases. 

```systemverilog
proto interface ProtoA #(
    param X: u32,
) {
    var a: logic;

    modport A {
        a: input,
    }
}

interface InterfaceA for ProtoA #(
    param X: u32,
) {
    var a: logic;

    modport A {
        a: input,
    }
}
```

So generic bound by interface may be sufficient instead of adding interface prototype.

```systemverilog
block rggen_connect_bit_field_if::<RIF: InterfaceA, FIF: InterfaceA, LSB: const, WIDTH: const> {
    assign  FIF.valid                 = RIF.valid;
    assign  FIF.read_mask             = RIF.read_mask[LSB+:WIDTH];
    assign  FIF.write_mask            = RIF.write_mask[LSB+:WIDTH];
    assign  FIF.write_data            = RIF.write_data[LSB+:WIDTH];
    assign  RIF.read_data[LSB+:WIDTH] = FIF.read_data;
    assign  RIF.value[LSB+:WIDTH]     = FIF.value;
}
```

taichi-ishitani, 2025-01-07:
The `prototype` generics boundary is to specify a module name which will be instantiated.
This notation may be confusable because its appearance is similar with the `prototype` boundary but thier functionality are different.
Therefore, I think other notation is needed like below.

```
block rggen_connect_bit_field_if::<RIF: inst InterfaceA, FIF: inst InterfaceA, LSB: const, WIDTH: const> {
    assign  FIF.valid                 = RIF.valid;
    assign  FIF.read_mask             = RIF.read_mask[LSB+:WIDTH];
    assign  FIF.write_mask            = RIF.write_mask[LSB+:WIDTH];
    assign  FIF.write_data            = RIF.write_data[LSB+:WIDTH];
    assign  RIF.read_data[LSB+:WIDTH] = FIF.read_data;
    assign  RIF.value[LSB+:WIDTH]     = FIF.value;
}
```

taichi-ishitani, 2025-01-08:
interface can have functions so I think interface prototype is meaningful.

```systemverilog
proto interface InterfaceA {
  function FuncA(a: input logic, b: input logic) -> logic;

  modport mp {
    FuncA: import,
  }
}
```

closed 2025-04-17 by dalance