veryl-lang/veryl #964: Can't use for loop variable in expression in if_reset statement

nananapo, 2024-09-13:
```sv
module main (
    clk: input clock,
    rst: input reset
)
{
    var r : logic<32,32>;
    always_ff{
        if_reset {
            for i : u32 in 0..32{
                r[i] = i;
            }
        }
    }
}
```

```
]8;;https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#invalid_reset_non_elaborative\invalid_reset_non_elaborative (link)]8;;\

  💥 Reset-value cannot be used because it is not evaluable at elaboration
  │ time
    ╭─[:9:24]
  8 │             for i: u32 in 0..32 {
  9 │                 r[i] = i;
    ·                        ┬
    ·                        ╰── Error location
 10 │             }
    ╰────
  help: 
```

nblei, 2024-09-13:
I think we do not have a mechanism to identify that the `i`  is a loop value through a fixed-range loop.

In the meantime, nananapo, you can use a generate loop which contains an `always_ff` block.

taichi-ishitani, 2024-09-13:
[image]

`genvar` is also treated as a variable.
It should be constant value.

taichi-ishitani, 2024-09-25:
For SystemVerilog, following code causes elaboration error because operand of part selection must be constant.

```systemverilog
for (int i = 0;i < W;++i) begin
  a[i:0] = i;
end
```

I think Veryl should treat a loop variable as a variable except when it is referred as a reset value.

dalance, 2024-09-25:
> I think Veryl should treat a loop variable as a variable except when it is referred as a reset value.

I think this rule is bit difficult, and may introduce ambiguous edge cases in the future. "`genvar` is constant, and loop variable is variable" is simple and sufficient?

taichi-ishitani, 2024-09-25:
Instead of this, how about disabling reset valeu check?

dalance, 2024-09-27:
Even if the check is removed, `a[i:0]` issue still remains.
I think changing `genvar` to constant is correct at least, and it resolves this issue by generate loop.

taichi-ishitani, 2024-09-27:
I read before an article that reducing number of procedural blocks such as `always` block is good technique to improve simulation performance.
Therefore, I think it's better to process an array within one `always` block but not multiple `always` blocks using `generate for` loop.

dalance, 2024-09-27:
OK. I'll merge #974, but leave this issue.

taichi-ishitani, 2024-09-29:
> Even if the check is removed, a[i:0] issue still remains.

Regular variables also cause this error so we need to introduce an additional check.

taichi-ishitani, 2024-10-03:
> > Even if the check is removed, a[i:0] issue still remains.
> 
> Regular variables also cause this error so we need to introduce an additional check. For this check, we can treat a loop variable as a regular variable and then don't need to treat it as a special variable.

I think this check can be done by #788.

dalance, 2024-10-03:
I think #788 is not so easy.
If `ConstantExpression: Expression` are added, how definition of `Select` is reasonable?
Probably the following is avaibable, but these many duplications occur whole syntax.

```
Select: LBracket Expression [ SelectOperator Expression ] RBracket;
ConstSelect: LBracket ConstExpression [ SelectOperator ConstExpression ] RBracket;
```

dalance, 2024-10-04:
I'm starting to feel that introducing`ConstExpression` is not so good idea.
The idea of `ConstExpression` means that whether expression is const or not can be determined syntactically.
Of course, sometimes it can be, but not always (e.g. reset value).
So the additional check for some `Expression` is reqruired even if `ConstExpression` is introduced.

I think inserting const check to `Expression` which is syntactically const is easy, and the place which is ambiguous is difficult regardless of introducing `ConstExpression`.
So I think `ConstExpression` doesn't has enough effect to reduce analyzer complexity.
Maybe inserting const check to each `Expression` is sufficient.

PR #1027 (2024-10-22): Allow to use loop variable as reset value
Closes #964
closed 2024-10-22 by dalance