veryl-lang/veryl #975: veryl-0.13.0 can'tt be built because of runaway builds

yurivict, 2024-09-27:
We have all amd64 builds failing because the Rust compiler doesn't produce an output, see the  [Ruust issue](https://github.com/rust-lang/rust/issues/130905).

Do you use any special options to build Veryl?

dalance, 2024-09-27:
If the following profile doesn't effect, compilation time increase extremely.
Is this option is used in the build?

```toml
[profile.release]
# Workaround for long time compilation
# https://github.com/rust-lang/rust/issues/106211
codegen-units = 1
```

https://github.com/rust-lang/rust/issues/106211 seems to be fixed, but there is another issue: https://github.com/rust-lang/rust/issues/129776.

dalance, 2024-09-27:
The `-C` options of the last command seems to contain `-C codegen-units=1`

```
-C opt-level=3 -C panic=abort -C linker-plugin-lto -C codegen-units=1 -C metadata=0edb5deee78252e7 -C extra-filename=-0edb5deee78252e7 -C linker=cc -C strip=debuginfo -C link-arg=-fstack-protector-strong`
```

The `-C` options of my local build log is below:

```
-C opt-level=3 -C embed-bitcode=no -C codegen-units=1 -C metadata=9d34c72853834718 -C extra-filename=-9d34c72853834718 -C linker=/usr/bin/cc -C strip=debuginfo
```

`-C linker-plugin-lto` is used in your environment only, and LTO is related to the compliation time issue.
So could you try to remove the `-C linker-plugin-lto`?

yurivict, 2024-09-27:
The FreeBSD port didn't add linker-plugin-lto, and such string doesn't appear in the veryl sources.

dalance, 2024-09-27:
Could you try the following profile?

```toml
[profile.release]
to = "off"
```

dalance, 2024-09-27:
LTO seems to be enabled by https://github.com/freebsd/freebsd-ports/blob/1779e67e6a02d99c1ccf885362c5fad50e4d50b5/Mk/Uses/cargo.mk#L144.

So If you define `WITHOUT_LTO` in `/cad/veryl/Makefile`, additional LTO is probably disabled.

yurivict, 2024-09-27:
Disabling LTO causes build to succeed.
This looks like a bug in Rust or in LLVM.

PR #976 (2024-09-29): Add `#[inline(never)]` to `expression*` as a workaround for long time compilation
Closes #975.

In some build configurations, Rust/LLVM luckily meets the inlining threshold, which inlined all expression functions to get a huge function, causing compile time problems.
closed 2024-09-29 by dalance