veryl-lang/veryl #987: genric function with same parameter cause errror

nananapo, 2024-10-03:
```sv
module ModuleA {
    function FuncA::<T: const> (
        a: input logic<T>,
    ) -> logic<T> {
        return a + 1;
    }

    let _a: logic<10> = FuncA::<10>(1);
    let _b: logic<10> = FuncA::<10>(1);
}
```
```
]8;;https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#invalid_factor\invalid_factor (link)]8;;\

  💥 __FuncA__10 of kind "generic instance" cannot be used as a factor in
  │ an expression
    ╭─[:9:25]
  8 │     let _a: logic<10> = FuncA::<10>(1);
  9 │     let _b: logic<10> = FuncA::<10>(1);
    ·                         ──┬──
    ·                           ╰── Error location
 10 │ }
    ╰────
  help: remove generic instance from expression
]8;;https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#call_non_function\call_non_function (link)]8;;\

  💥 Calling non-function symbol "__FuncA__10" which has kind "generic
  │ instance"
    ╭─[:9:25]
  8 │     let _a: logic<10> = FuncA::<10>(1);
  9 │     let _b: logic<10> = FuncA::<10>(1);
    ·                         ──┬──
    ·                           ╰── Error location
 10 │ }
    ╰────
  help: remove call to non-function symbol
```

nananapo, 2024-10-04:
generic module/interface instantiation also failed😭
generic package is ok.

dalance, 2024-10-04:
By #993, function, module and interface are fixed.

PR #993 (2024-10-04): Fix error by repeated generics usage
Closes #987
closed 2024-10-04 by dalance