veryl-lang/veryl #990: Can't resolve imported const in generic interface modport

nananapo, 2024-10-03:
version: https://github.com/veryl-lang/veryl/commit/978fdb0ab7453bf206e3c6a381eca7c5f5fe678c

top.veryl
```sv
import pkg::*;

module Top {
    inst m : myif::<XLEN>;
}
```
pkg.veryl
```sv
package pkg {
    const XLEN :u32 = 32;
}
```
myif.veryl
```sv
interface myif::<WIDTH: const> {
    var data : logic<WIDTH>;
    modport master {
        data : output,
    }
}
```

```
Error: undefined_identifier (link)

  × XLEN is undefined
  [Failed to read contents for label `Error location` (offset: 54, length: 5): OutOfBounds]
  help: 
```

Using pkg::XLEN instead of XLEN compiles successfully.

taichi-ishitani, 2024-10-03:
I think this is the same issue with #754 .

dalance, 2024-10-04:
In #754, type defined in module can't be refered from outside of the module.
But in this case, the `XLEN` can be refered by `pkg::XLEN`, so Veryl compiler can add `pkg::` path safely.

PR #1034 (2024-10-23): Resolve imported generic arguments
Closes #990