veryl-lang/veryl #1002: "Verylup setup" doesn't support offline install

captainliuy, 2024-10-09:
The 'verylup setup' command requires an internet connection, but my office development environment does not support it. Please add a option for offline installation.

captainliuy, 2024-10-09:
Download the latest installation package directly from GitHub, copy it to the  development environment, and extract it to the directory where verylup is located.

dalance, 2024-10-09:
Yes. You are right.
But the way disables some verylup features like version switching.
So offline installation support may be useful at some cases.

For example, is the following way useful  for you?

```
$ verylup setup --offline --pkg veryl-x86_64-linux.zip
$ veryl --version
veryl 0.13.0
$ verylup install 0.11.0 --pkg veryl-x86_64-linux.zip
$ veryl +0.11.0 --version
veryl 0.11.0
```

captainliuy, 2024-10-10:
> Yes. You are right. But the way disables some verylup features like version switching. So offline installation support may be useful at some cases.
>
> For example, is the following way useful for [...]

Yes, that will work perfectly for me.

dalance, 2024-10-11:
I released Verylup 0.1.3 with offline installation support.

closed 2024-10-11 by dalance