veryl-lang/veryl #1005: variable of generic interface instance is inaccessible

nananapo, 2024-10-10:
```sv
interface InterfaceA::<C: const> {
    var value: logic<C>;
    modport master {
        value: input,
    }
}

module Top(
    port_i : input InterfaceA::<2>,
    modport_i : input InterfaceA::<3>
) {
    inst inst_i: InterfaceA::<1>;

    always_comb {
        port_i.value = 0; // ok
        modport_i.value = 0; // ok
        inst_i.value = 0; // ng
    }
}
```

```
Error:   × veryl check failed

Error: unknown_member (link)

  × "inst_i" doesn't have member "value"
    ╭─[/home/kanataso/Documents/bluecore/aaa/main.veryl:17:9]
 16 │         modport_i.value = 0; // ok
 17 │         inst_i.value = 0; // ng
    ·         ──────┬─────
    ·               ╰── Error location
 18 │     }
    ╰────
  help: 
```

taichi-ishitani, 2024-10-10:
An interface port cannot have port direction so Veryl should report an error for this code but does not.

```
    port_i : input InterfaceA::<2>,
    modport_i : input InterfaceA::<3>
```

This is another issue.

nananapo, 2024-10-10:
```sv
interface InterfaceA::<C: const> {
    var value: logic<C>;
    modport master {
        value: output,
    }
}

module Top(
    modport_i : modport InterfaceA::<3>::master
) {
    inst inst_i: InterfaceA::<1>;

    always_comb {
        modport_i.value = 0; // ok
        inst_i.value = 0; // ng
    }
}
```

fixed example code :(
error message is same

PR #1009 (2024-10-10): Fix member access failure of generic instance
Closes #1005
closed 2024-10-10 by dalance