veryl-lang/veryl #1006: No error is reported for interface port with port direction

taichi-ishitani, 2024-10-10:
An interface port cannot have port direction so Veryl should report an error for this code but does not.

```
    port_i : input InterfaceA::<2>,
    modport_i : input InterfaceA::<3>
```

_Originally posted by @taichi-ishitani in https://github.com/veryl-lang/veryl/issues/1005#issuecomment-2403995006_

taichi-ishitani, 2024-10-11:
If the given interface has no generic parameters then the `type mismatch` error is reported.
[image]

Additonal investigation is needed.

dalance, 2024-10-11:
May the port direction be mistake? How about `modport InterfaceN1::mp`?

taichi-ishitani, 2024-10-11:
> If the given interface has no generic parameters then the `type mismatch` error is reported. [...]

This check is executed before resolving generic parameters so `type mismatch` error is not reported for an interface port generic parameters.
The checker should be moved after resolving generic parameters.

PR #1011 (2024-10-11): Execute `type check` afeter resolving generics
The `type check` checker requires resolved generic instanciations but, currently, it is executed before resolving them.
This is cause of #1006 and #1007.

Therefore, the checker should be executed after resolving generic instanciations.

fix #1006
fix #1007