veryl-lang/veryl #1028: Refer a variable before assigning it

taichi-ishitani, 2024-10-22:
A variable should be assigned before it is referred to get correct simulation result.

```systemverilog
// OK
always_comb {
  a = 1;
  b = a;
}

// NG
always_comb {
  d = c;
  c = 1;
}
```

Veryl should report an error fo this kind of code but currently not.
[image]

nananapo, 2024-10-22:
Instead of reporting error, I think it's better to sort automatically by Veryl compiler. (or create option)
Reordering manually to make the simulation work correctly is only for simulation and is not a essential task.

nananapo, 2024-10-22:
hmm....
reordering maybe difficult when there are if or case in always_comb😞

closed 2024-11-05 by dalance