veryl-lang/veryl #1046: Add license file to new project

taichi-ishitani, 2024-10-31:
Add license file automatically when creating a new project by using `veryl new` command.

dalance, 2025-03-26:
Is there any reference implemetation which have such feature?

taichi-ishitani, 2025-03-26:
> Is there any reference implemetation which have such feature?

How about `bundler` gem?
The `bundle` command copy the MIT license file and add license information to the README file when initializing Ruby project with the `--mit` option.
https://github.com/veryl-lang/veryl/issues/1046#issuecomment-2753387797
https://github.com/rubygems/rubygems/blob/557237b70c8f22e016742c046dac6531dbc95be1/bundler/lib/bundler/templates/newgem/LICENSE.txt.tt
https://github.com/rubygems/rubygems/blob/557237b70c8f22e016742c046dac6531dbc95be1/bundler/lib/bundler/templates/newgem/README.md.tt#L38-L43

dalance, 2025-03-26:
Do you know how `bundle` gets author's information which should be embedded in license file?

taichi-ishitani, 2025-03-26:
It seems that bunder gets author info from `git config`.
https://github.com/rubygems/rubygems/blob/557237b70c8f22e016742c046dac6531dbc95be1/bundler/lib/bundler/cli/gem.rb#L48-L50