veryl-lang/veryl #1054: Charactor wrapping modport items

taichi-ishitani, 2024-11-06:
Currently, modport items are wrapped by curly brackets.

```systemverilog
modport master {
  ready:   input,
  valid:   output,
  payload: output
}
```

On the other hand, port declarations are wrapped by parentheses.

```systemverilog
module foo (
  i_clk: input  clock,
  i_rst: input  reset,
  i_d:   input  logic,
  o_d:   output logic
) {}
```

Both of modport item and port declaration is to define a port. So I think `modport` items should be wrapped by parentheses to align with port declarations.

dalance, 2024-11-06:
I think modport is a composite data type definition like struct.
Additionally most of declarations have `{}` as a body.
So I think the usage of `{}` is consistent.

```systemverilog
modport A {}
struct A {}
always_comb {}
module A {}
interface A {}
```

taichi-ishitani, 2024-11-06:
OK, I understood.

closed 2024-11-06 by taichi-ishitani