veryl-lang/veryl #1102: Wrong filelist order

taichi-ishitani, 2024-11-27:
If a module includes a reference to an item declared within a package using the scope resolution `::` the filelist order should be

1. pacakge
2. module

But actual order is swapped.

Following is reproduce procedure.

1. Checkout https://github.com/rggen/rggen-veryl-rtl repository
2. Run `veryl build` on the repository

`rggen_axi4lite_skid_buffer` includes such reference so it should follow `rggen_rtl_pkg.sv` on the generated filelist.
https://github.com/rggen/rggen-veryl-rtl/blob/7c3c781e0ad3d5d2948711423758e120d36d8bc7/rggen_axi4lite_skid_buffer.veryl#L11

Actual generated filelist is below and order between `rggen_rtl_pkg` and `rggen_axi4lite_skid_buffer` is swapped.

```
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_axi4lite_if.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_wishbone_if.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_axi4lite_skid_buffer.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_apb_if.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_mux.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_rtl_pkg.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_bit_field_if.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_bit_field_w01trg.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_bit_field.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_register_if.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_address_decoder.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_register_common.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_indirect_register.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_default_register.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_bus_if.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_wishbone_bridge.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_axi4lite_bridge.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_adapter_common.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_wishbone_adapter.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_axi4lite_adapter.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_apb_adapter.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_apb_bridge.sv
/home/taichi/workspace/rggen/rggen-veryl-rtl/rggen_external_register.sv
```

dalance, 2024-11-27:
This bug seems to be the same as #1070.

taichi-ishitani, 2024-11-27:
> This bug seems to be the same as #1070.

Yes, it's the same issue.

closed 2024-11-27 by taichi-ishitani