veryl-lang/veryl #1142: resolving msb of struct member is failed

nananapo, 2024-12-13:
```sv
module TopModule {
    struct A {
        a: logic<6>,
    }
    let _a: A        = 0;
    let _b: logic<5> = _a.a[msb:1];
}
```

```
]8;;https://doc.veryl-lang.org/book/07_appendix/02_semantic_error.html#unknown_msb\unknown_msb (link)]8;;\

  💥 resolving msb is failed
   ╭─[:6:29]
 5 │     let _a: A        = 0;
 6 │     let _b: logic<5> = _a.a[msb:1];
   ·                             ─┬─
   ·                              ╰── Error location
 7 │ }
   ╰────
  help: 
```

taichi-ishitani, 2024-12-17:
Veryl should report `unknown msb` error for interface instance like below but does not.

```
    interface InterfaceA #(
        param W: u32: 2
    ){
        var a: logic<W>;
        modport mp {
            a: input
        }
    }
    module ModuleA {
        inst if_a: InterfaceA;
        let a: logic = if_a.a[msb];
    }
```

[image]

taichi-ishitani, 2024-12-17:
Veryl will generate illegal SV for the case where a struct is defied outside of the declaration scope like below.

Original Veryl
```systemverilog
package Package28A::<W: const> {
    struct StructA {
        a: logic<W>,
    }
}

package Package28B {
    const B: u32 = 2;
}

package Package28C {
    const W: u32    = 2;
    const C: bit<W> = 0;
}

module ModuleB {
    var a: Package28A::<Package28B::B>::StructA;
    assign a.a = 0;

    let _y: logic = a.a[msb];
    let _z: logic = Package28C::C[msb];
}
```

Generated SV
```systemverilog
package veryl_testcase___Package28A__Package28B_B;
    typedef struct packed {
        logic [veryl_testcase_Package28B::B-1:0] a;
    } StructA;
endpackage

package veryl_testcase_Package28B;
    localparam int unsigned B = 2;
endpackage

package veryl_testcase_Package28C;
    localparam int unsigned         W = 2;
    localparam bit          [W-1:0] C = 0;
endpackage

module veryl_testcase_ModuleB;
    veryl_testcase___Package28A__Package28B_B::StructA a  ;
    always_comb a.a = 0;

    logic _y;
    always_comb _y = a.a[(() - 1)]; // Width part is empty
    logic _z;
    always_comb _z = veryl_testcase_Package28C::C[((W) - 1)]; // W is not defined in this scope
endmodule
```

I think using `$bits` system function is better than evaluating the `expression`.

taichi-ishitani, 2024-12-17:
> I think using $bits system function is better than evaluating the expression like below.

`$bits` returns total amount of bits of the given type or expression so it is not suitable to get the array size.

```
logic [1:0][2:0][3:0] a;
$bits(a): 24
$bits(a[0]): 12
$bits(a[0][0]): 4
```

SV also supports `$size` system function which returns the array size.
[image]

DC, Vivado and also Verilator support this system function so we can use it to calcurate `msb` position.

taichi-ishitani, 2024-12-17:
> DC, Vivado and also Verilator support this system function so we can use it to calcurate msb position.

When this causes problems we need to calculate `msb` position from the variable declaration during emitting SV source files.

PR #1154 (2024-12-19): Refactor `msb` check and emittion
This PR is to fix following two issues of `msb` feature.

* `msb` cannot be used on struct/union member, enum and paramter
* Wrong msb position calcuration for a type defined in outside of the current scope
  * https://github.com/veryl-lang/veryl/issues/1142#issuecomment-2547700428
  * use `$size` and `$bits` system function instead of evaluating the size declaration

fix #1142
closed 2024-12-19 by dalance