veryl-lang/veryl #1155: Avoid `Warning-DECLFILENAME` warning reported by Verilator

taichi-ishitani, 2024-12-18:
Verilator reports a `Warning-DECLFILENAME` warning because it checks consistency between module name and file name.
https://qiita.com/mikecat_mixc/items/cf98dffa4891f00f132f#hello-world

I think it would be nice to add the `/* verilator lint_off DECLFILENAME*/` and `/* verilator lint_on DECLFILENAME*/` pragma comments to head and end of generated SV files to avoid this warning.

dalance, 2024-12-20:
I think such tool-specific features should be optional at least.
And the need is questionable because there are many workarounds like below:

* Use `omit_project_prefix = true`
* Don't add `-Wall` to Verilator
* Add `-Wno-DECLFILENAME` to Verilator

taichi-ishitani, 2024-12-20:
OK, I understand. I'm closing this ticket.

closed 2024-12-20 by taichi-ishitani