veryl-lang/veryl #1178: Default argument of function

dalance, 2025-01-03:
#1164 introduced default argument of function too, but I think more consideration is required.

* Ambiguous position of default argument

In the following case, calling `FuncA(0, 0)` is ambiguous. It may be `i_a = 1, i_b = 0, i_c = 0` or `i_a = 0, i_b = 1, i_c = 0`.
To prevent this issue, the restriction that default arguments should be placed at last may be required.

```
function FuncA (
    i_a: input logic = 1,
    i_b: input logic = 1,
    i_c: input logic,
) {}
```

* Lower explicitness

At module instantiation, port names are shown, and it is clear which ports are omitted.
But at function, argument names are not shown, so explicitness is lower than module instantiation.
Forbidding default argument may be reasonable as the same as Go / Rust.

dalance, 2025-01-03:
@taichi-ishitani I think temporarily disabling this feature for v0.13.4 is better. How about it?

taichi-ishitani, 2025-01-03:
> @taichi-ishitani I think temporarily disabling this feature for v0.13.4 is better. How about it?

I agree with you. I will open a PR for this.

taichi-ishitani, 2025-01-03:
> > @taichi-ishitani I think temporarily disabling this feature for v0.13.4 is better. How about it?
> 
> I agree with you. I will open a PR for this.

I opened PR #1181 for this.