veryl-lang/veryl #1187: std test failure with VCS

dalance, 2025-01-07:
`test_std_types_pkg` with VCS is failed like below:

```
[INFO ]    Compiling test (test_std_types_pkg)
[INFO ]    Executing test (test_std_types_pkg)
error detected
error detected
error detected
error detected
[ERROR]       Failed test (test_std_types_pkg)
[ERROR]    Completed tests : 0 passed, 1 failed
```

The failed point is 

https://github.com/veryl-lang/veryl/blob/bbc5d9a1d8c9f83bb6f8861d8c6888bc9e2afb9d/crates/std/veryl/src/pkg/types.veryl#L86-L88

`assert(VARIABLE == '0)` is OK.

taichi-ishitani, 2025-01-08:
This test is for unsigned bit vector types so this assertion `(VARIABLE == '0)` should be passed.

dalance, 2025-01-08:
Should we use `==` instead of `===` in this case?

taichi-ishitani, 2025-01-08:
This assertion checks if `VARIABLE` does not contain x/z so we should use `===` operator.

taichi-ishitani, 2025-01-08:
> Do you think this issue is a bug of VCS?

Yes, I think so.

taichi-ishitani, 2025-01-08:
In facto, Xcelium, Questa and Riviera report no errors for this code.

dalance, 2025-01-09:
Close by #1195

closed 2025-01-09 by dalance