veryl-lang/veryl #1205: Dependency specification without version constraint

taichi-ishitani, 2025-01-17:
In case of specifying dependencies on local storage, such dependencies may be under development and version control is not suitable for them.
I think it would be nice to allow to specify such dependencies without version constraint and import existing source files in the dependencies as is.

dalance, 2025-01-17:
I thought an actual workflow like below:

* develop a project using local path dependency
* publish the dependency project by new version
* publish the project by new version

In this case, both local path and version is required.
So the following specification is probably reasonable.

```toml
[dependencies]
"https://github.com/veryl-lang/sample" = { version = "0.1.0", path = "../../sample" }
```

This means "If `path` exists, it is used as is, and if not, `version` is used from the source repository".

dalance, 2025-01-20:
I implemented this feature at #1206.
Now the following path specification can be used.

```
[dependencies]
"https://github.com/veryl-lang/sample" = { version = "0.1.0", path = "../../sample" }
"../../sample"                         = { version = "0.1.0", path = "../../sample" }
```

In this PR, there are the following restrictions:

* left-hand side URL/path is mandatory. It should be valid git repository.
* `version` is mandatory. It should be published.

These restrictions are required to keep consistency with the existing lockfile system.
How about it?

PR #1206 (2025-01-23): Support to override dependencies with local path
Closes #1205
closed 2025-01-23 by dalance