veryl-lang/veryl #1223: [Feature] `pacakge` generic boundary

taichi-ishitani, 2025-02-04:
Some IPs use a lot of typedefs defined by its structued config parameters.
For example:
* https://github.com/pezy-computing/pzbcm/blob/985e7147ab42682e62f32604433090b3be5f8482/pzcorebus_common/pzcorebus_if_internal_macros.svh#L10-L29
* https://github.com/pezy-computing/pzbcm/blob/985e7147ab42682e62f32604433090b3be5f8482/pzcorebus_common/pzcorebus_if.sv#L18

For this kind of useage, the config parameter and typedefs should be handled together.
I'd like to introduce the `package` generic boundary to allow to specify a generic package as a generic argument.

Example code:

```systemverilog
package foo_pkg::<AW: const, DW: const> {
  const ADDRESS_WIDTH: u32 = AW;
  const DATA_WIDTH:    u32 = DW;

  type address_t = logic<ADDRESS_WIDTH>;
  type data_t    = logic<DATA_WIDTH>;
}

module foo_top::<PKG: foo_pkg> (
  i_address: input PKG::address_t,
  i_data:    input PKG::data_t,
) {
  import PKG::*;

  var address: address_t;
}

inst u_foo: foo_top::<foo_pkg::<16, 32>>(
  i_address: address,
  i_data:    data,
);
```

taichi-ishitani, 2025-02-19:
For the above sample, `foo_pkg` has only 2 generic parameters but an actual packages may have more generic parameters. I think it is very troublesome to write such generic instance everywhere it is used.

To avoid this, I think it would be nice to create an alias to such generic package instance.

closed 2025-02-27 by dalance

nblei, 2025-03-30:
@taichi-ishitani This is pretty cool.

Is there a way to alias a specific package instance?

E.g., I have

```systemverilog
package fp_pkg::<Exponent: const, Mantissa: const> {
    struct fp {
        sign    : logic          ,
        exp     : logic<Exponent>,
        mantissa: logic<Mantissa>,
    }

    struct cfp {
        real: fp,
        imag: fp,
    }
}
```

Thus, I would like something like `let fp16_pkg = fp_pkg::<5, 10>`.

taichi-ishitani, 2025-03-30:
You can create an alias to a package like below.

```systemverilog
alias package fp16_pkg = fp_pkg<5, 10>;
```

refs: https://github.com/veryl-lang/veryl/issues/1296