veryl-lang/veryl #1224: [Bug] Module is marked as duplicated after moving the source file

Hamster5295, 2025-02-04:
Hello,   
I'm currently using Veryl Language Server with Veryly VSCode extension.  

It occurred that after moving a source file from a directory to another, the module and all its members (the `logic`s and `const`s) are marked as duplicated.  `Semantic Error: ModuleA is duplicated` as an instance.  

I'm working on VSCode with Veryl extension version `0.13.5`.  
My system is Windows 11 24H2.

Hamster5295, 2025-02-04:
Deleting the source file of `ModuleA` then create another file in different directory with another module named `ModuleA` also triggers it.

dalance, 2025-02-05:
Thank you for your report!
Now Veryl VSCode extension doesn't support to handle file operation like rename and delete.
I'll try to add it.

Hamster5295, 2025-02-05:
I'm currently working on it, trying to register `did_rename_file` events for language server and remove the old file uri from `veryl_analyser::symbol_table` and `veryl_analyzer::namespace_table` when rename operations occur. However, it failed to solve the bug.  

My code goes like this:
```rust
/**
* server.rs
*/

// this function is called when backend.rs send a message in did_rename_file callback, removing the old file url
// through logging the `drop` function is guaranteed to have been executed, and the path is correct.
// However, this does not fix the case. 
// Is there somewhere else that's adding those symbols to the table, or I might have misunderstood the whole logic?
fn on_remove(&mut self, path: String) {
    if let Ok(url) = Url::from_str(&path) {
        if let Ok(path) = url.to_file_path() {
            if let Some(path) =
                resource_table::get_path_id(Path::new(&path).to_path_buf())
            {
                symbol_table::drop(path);
                namespace_table::drop(path);
            }
        }
    }
}
```

dalance, 2025-02-05:
It is just a code which I imaged.
If you want to handle `did_rename_file`, not only dropping old path but also adding new path is probably necessary.

The following code may be helpful to add new path.

https://github.com/veryl-lang/veryl/blob/eddc363e7b9d7604714e215454d263c3a1db4444/crates/languageserver/src/server.rs#L207-L215

Hamster5295, 2025-02-05:
get it, I'll try later 🐱

Hamster5295, 2025-02-05:
Unfortunately, it doesn't solve the case.  

[image]
As is shown above the `did_open` triggers right before `did_rename`, which has already dispatched a background task for analyzing.  
From the output it can be figured out that the correct path of the renamed file was passed to the task when `did_open` is triggered.   
Thus, after adding the code above when `did_rename` is triggered, the same code base is analyzed again, while the bug persists.

```txt
[Info  - 17:47:56] did_open
[Info  - 17:47:56] did_rename

...Analysis of dependencies...

[Info  - 17:47:56] background_analyze: \\?\E:\path\to\my\testing\project\top.veryl

...Analysis of dependencies...

[Info  - 17:47:56] background_analyze: \\?\E:\path\to\my\testing\project\top.veryl
```

Maybe it's time to dig deeper into the analyzer...

Hamster5295, 2025-02-05:
I've pushed my current modifications to my fork of veryl. Check it out if you need further information and testing :D
[server.rs](https://github.com/Hamster5295/veryl/blob/master/crates/languageserver/src/server.rs)

dalance, 2025-02-05:
How about using [will_rename_files](https://docs.rs/tower-lsp/latest/tower_lsp/trait.LanguageServer.html#method.will_rename_files)? 
It seems to be sent before the actual renaming, so I image the following sequence.

will_rename_files
drop old_file
did_open 
analyze new_file
did_rename

If there is the case that did_open is not sent, analyze new_file after did_rename may be necessary.

Hamster5295, 2025-02-05:
> How about using [will_rename_files](https://docs.rs/tower-lsp/latest/tower_lsp/trait.LanguageServer.html#method.will_rename_files)? It seems to be sent before the actual renaming, so I image the following sequence.

It worked out great! I'll work on to clean up and do some further testings.

Hamster5295, 2025-02-05:
> If there is the case that did_open is not sent, analyze new_file after did_rename may be necessary.

Indeed, but it seems that `metadata.paths::<&str>(&[], true)` still gets the paths after renaming.  
```txt
[Info  - 20:21:00] did_rename
[Info  - 20:21:00] Symbols from e\path\to\my\test\project\src\top2.veryl are dropped!  

......

[Info  - 20:21:00] background_analyze: e\path\to\my\test\project\src\top.veryl
```  

This might be an undefined behaviour however, as my `will_rename_files` goes like:
```rust
    async fn will_rename_files(&self, params: RenameFilesParams) -> Result<Option<WorkspaceEdit>> {
        self.client
            .log_message(MessageType::INFO, "will_rename_files")
            .await;

        for file in params.files {
            self.send(MsgToServer::WillRenameFile {
                old_url: file.old_uri,
                new_url: file.new_uri,
            })
            .await;
        }
        Ok(None)
    }
```
which is likely to introduces a race condition between our `server.rs` and VSCode Language Server Host.  
I'll explicitly dispatch a new background analyze task at `did_rename_files` to avoid that.

PR #1226 (2025-02-06): [Fix] Basic Support for File Renaming and Deleting
Closes #1224
closed 2025-02-06 by dalance