veryl-lang/veryl #1246: [Featur] Modport which will be expanded into each Verilog ports

taichi-ishitani, 2025-02-13:
We're using text macros to expand a modport into each Verilog ports and to merge Verilog ports into an interface. In SystmeVerilog, we have no choise except for using text macro.

https://github.com/pezy-computing/pzbcm/blob/985e7147ab42682e62f32604433090b3be5f8482/pzcorebus_common/pzcorebus_macros.svh#L24-L30

To handle this kind of case, I'd like to introduce a modport which will be expanded into each Verilog ports during emitting SV code.

For example:

Veryl code
```systemverilog
interface foo_if::<W: const> {
  var ready: logic;
  var valid: logic;
  var data : logic<W>;

  modport slave {
    ready: output,
    valid: input ,
    data : input ,
  }
}

module foo (
  foo_0_if: modport foo_if::<32>::slave
) {

}
```

Generated SV code
```systemverilog
module foo (
  output logic          foo_0_if_ready,
  input  logic          foo_0_if_valid,
  input  logic [32-1:0] foo_0_if_data
);
  foo_if_32 foo_0_if();
  always_comb foo_0_if_ready = foo_0_if.ready;
  always_comb foo_0_if.valid = foo_0_if_valid;
  always_comb foo_0_if.data  = foo_0_if_data;
endmodule
```

dalance, 2025-02-28:
This feature seems to be good for bridge module between Verilog and SystemVerilog.

```systemverilog
#[expand(modport)]
module SvModuleForVerilog (
    foo_0_if: modport foo_if::<32>::slave
) {

    inst u: SvModule (
        foo_0_if,
    );
}

module VerilogModuleForSv (
    foo_0_if: modport foo_if::<32>::slave
) {

    #[expand(modport)]
    inst u: VerilogModule (
        foo_0_if,
    );
}
```

Additionally, other patterns like `#[expand(struct)]`, `#[expand(modport, struct)]` can be considered.

taichi-ishitani, 2025-03-11:
For ease to emit SV code, expandable interfaces should have no parameters and constants. They should depend on packages and literals only.

taichi-ishitani, 2025-04-03:
I'm working on this.