veryl-lang/veryl #1257: Bit width check policy

dalance, 2025-02-19:
Now bit width checking becomes ready to be implemented.
So we should decide what types of bit width difference are checked.

For example, the following code is commonly used as a wrap around counter.

```systemverilog
var cnt: logic<10>;

always_ff {
    cnt += 1;
}
```

From a point of view of most strict checking, `cnt += 1` should be detected as bit width error because it causes carry overflow.
But if these cases get error or warning, we feel troublesome and disable the checking.
So we should explore tradeoff between convenience and safety.

dalance, 2026-04-03:
At the first step, bit width mismatch should be checked assuming the current bit width calcration ( as the same as SV ).

```systemverilog
var a: logic<8>;
var b: logic<16>;

a = 10000;    // NG: 32bit -> 8bit
a = a + 8'd1; // OK: 8bit -> 8bit
a = b;        // NG: 16bit -> 8bit
b = a;        // OK: 8bit -> 16bit
```

There is a problem around base-less number.

```systemverilog
var a: logic<8>;

a += 1; // OK / NG ?
```

According to SV standard, base-less number is 32bit, so the above case is `32bit -> 8bit`.
However, the above case is clearly safety.
This is related to #2068, if `1` means `1'd1`, this issue is resolved.

taichi-ishitani, 2026-04-03:
> I prefer a strict checking

I think such check can be implemented in  the std function.
Can you open an issue for this?

omasanori, 2026-04-03:
> > I prefer a strict checking
> 
> I think such check can be implemented in the std function. Can you open an issue for this?

I meant the following cases raised by @dalance:

```systemverilog
var a: logic<8>;
var b: logic<16>;

a = 10000;    // NG: 32bit -> 8bit
a = a + 8'd1; // OK: 8bit -> 8bit
a = b;        // NG: 16bit -> 8bit
b = a;        // OK: 8bit -> 16bit
```

Not this one:

```systemverilog
var cnt: logic<10>;

always_ff {
    cnt += 1;
}
```

I am fine with that as long as the behavior is sound and guaranteed.

taichi-ishitani, 2026-04-03:
It is bother to write number literal of which width is same with `a` if its width is parameterized.
For such case, number literal with contextual width is useful, I think.
How about using normal number literal (e.g. 10 0xab 0b11) for this?