veryl-lang/veryl #1287: Native simulator

dalance, 2025-02-26:
Now all symbol information is already gathered, clock / reset is identified syntactically, and expression evaluator is ready.
So I think implementing Veryl native simulator is not so difficult.

The aim of the simulator is not simulation performance, but quick response.
All other solutions like Verilator and commercial simulator take long   compilation time. 
If simulation of the first one cycle complete in a second, it is probably valuable.

perlindgren, 2025-03-25:
I was checking the `MEMO.md` but its all in Japanese (and I don't trust automatic translation to correctly describe your intent with the describing text). Could you add an translation to English?

dalance, 2025-03-25:
I added English version.
https://github.com/veryl-lang/veryl/blob/master/crates/simulator/MEMO_en.md

dalance, 2026-02-03:
In #2179, I tried to implement simulator as interpreter of analyzer IR, but the performance was not so good.
Many changes to the IR are necessary to improve performance, but it causes unexpected complexity.

So I'm tring another implementation at #2197. It introduces simulator IR optimized for simulation.

closed 2026-03-16 by dalance