veryl-lang/veryl #1296: [Feature] Alias to package

taichi-ishitani, 2025-02-27:
It is very troublesome to write a package name which has a lot of generic parameters down everywhere is is used.
Alias to such package name can pretend this kind of case.

Example:

```systemverilog
pacakge foo_pkg::<A: const, B: const, C: const> {}
pacakge bar_pkg = foo_pkg::<1, 2, 3>;

module baz_module::<PKG: foo_pkg> {}

inst u baz_module::<bar_pkg>;
```

refs: https://github.com/veryl-lang/veryl/issues/1223#issuecomment-2667319385

taichi-ishitani, 2025-02-27:
I think it would be nice that the emitted package name is also replaced with alias name like below.

* `foo_pkg___1__2__3` -> `bar_pkg`
* `baz_module___foo_pkg___1__2__3` -> `baz_module___bar_pkg`

taichi-ishitani, 2025-02-27:
@dalance ,

Which syntax do you prefer?

* Introduce a new keyword

```systemverilog
alias bar_pkg = foo_pkg::<1, 2, 3>;
```

* Reuse the existing keyword

```systemverilog
package bar_pkg = foo_pkg::<1, 2, 3>;
```

dalance, 2025-02-27:
I prefer `package bar_pkg = foo_pkg::<1, 2, 3>;` because it shows that it is package definition more clearly.
I think `module` and `interface` may be useful too.

dalance, 2025-03-06:
`package bar_pkg = foo_pkg::<1, 2, 3>;` seems to increse syntax complexity.
So I think the following syntax is better.

```systemverilog
alias package bar_pkg = foo_pkg::<1, 2, 3>;
```

closed 2025-03-06 by dalance