veryl-lang/veryl #1300: Specify default clock / reset

dalance, 2025-02-27:
Sometimes, there are some clocks in a module, but only a specific clock is used in always_ff.

```systemverilog
let gated_clock: clock = if clock_enable { i_clk } else { 0 }; 

// only gated_clock is used 
always_ff ...
```

So specifing default clock/reset may be useful.

taichi-ishitani, 2025-03-06:
How about the syntax below?

```systemverilog
default clock i_clk_a;
default reset i_rst_a;
```

I think that a default clock declaration in a generate block should be prohibited because such case should be separated into a module.
What do you think?

dalance, 2025-03-11:
> I think that a default clock declaration in a generate block should be prohibited because such case should be separated into a module.

I think so too.

About syntax, I have a concern about the following case.

```systemverilog
module ModuleA (
    i_clk: input clock,

    ....

) {
    // far from i_clk declaration
    default clock i_clk;
}
```

How about type modifier or attribute?

```systemverilog
    i_clk: input default clock,

    #[default]
    i_clk: input clock,
```

taichi-ishitani, 2025-03-11:
> How about type modifier or attribute?

I prefer modifier.

taichi-ishitani, 2025-03-11:
I'm working on this.

closed 2025-03-12 by dalance