veryl-lang/veryl #1314: Cocotb abnormal exit is ignored

clkbug, 2025-03-02:
Hi,

The below code expects the test to fail.
However, veryl command actually reports a success.

code:
```systemverilog
module A (
    a: input logic<1>,
) {
    let b: logic<2> = a; // this line should cause a mismatch-width error
}

#[test(test, A)]
embed (cocotb) py{{{
import cocotb

@cocotb.test()
async def test(dut):
    assert False, "this test is expected to fail"
}}}
```

command: `veryl test`

output:
```
[INFO ]    Executing test (test)
[INFO ]    Succeeded test (test)
[INFO ]    Completed tests : 1 passed, 0 failed
```

As far as I can tell, this is due to an error in the build phase by Verilator, and `runner.py` exits with exit code 1, which is ignored.

dalance, 2025-03-03:
I added exit code check and stderr parser for cocotb.

```
[INFO ]    Executing test (test)
/tmp/.tmpShr0Br/runner.py:3: UserWarning: Python runners and associated APIs are an experimental feature and subject to change.
  import cocotb.runner
%Warning-WIDTHEXPAND: /home/hatta/work/repos/veryl/temp/test.sv:5:36: Operator ASSIGN expects 2 bits on the Assign RHS, but Assign RHS's VARREF 'a' generates 1 bits.
                                                                    : ... note: In instance 'temp_A'
    5 |     logic [2-1:0] b; always_comb b = a;
      |                                    ^
                      ... For warning description see https://verilator.org/warn/WIDTHEXPAND?v=5.024
                      ... Use "/* verilator lint_off WIDTHEXPAND */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
Process 'perl' terminated with error 1
[ERROR]       cocotb failed by error code exit status: 1
[ERROR]       Failed test (test)
[ERROR]    Completed tests : 0 passed, 1 failed
```

closed 2025-03-03 by dalance

clkbug, 2025-03-03:
Thanks for the quick fix!

PR #1316 (2025-03-03): Handle cocotb execute error
Closes #1314