veryl-lang/veryl #1337: Build failure when veryl-parser is used in proc-macro

dalance, 2025-03-05:
The following crate is failed to build.

```Cargo.toml
[package]
name = "test"
version = "0.1.0"
edition = "2021"

[lib]
proc-macro = true

[dependencies]
veryl-parser = "0.14.1"
```

```
error: failed to run custom build command for `scnr v0.8.0`

Caused by:
  process didn't exit successfully: `/home/hatta/work/repos/playground/test97/target/debug/build/scnr-bd8da6f8917803f2/build-script-build` (exit status: 101)
  --- stderr

  thread 'main' panicked at /home/hatta/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/scnr-0.8.0/build.rs:6:9:
  The features `default` and `regex_automata` are mutually exclusive.
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...
error: failed to run custom build command for `parol_runtime v2.2.1`

Caused by:
  process didn't exit successfully: `/home/hatta/work/repos/playground/test97/target/debug/build/parol_runtime-ee280bffdcbfbb3f/build-script-build` (exit status: 101)
  --- stderr

  thread 'main' panicked at /home/hatta/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/parol_runtime-2.2.1/build.rs:6:9:
  The features `default` and `regex_automata` are mutually exclusive.
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

In the proc-macro, both `dependencies` and `build-dependencies` are enabled, so `parol_runtime` with `regex_automata` from `dependencies` and `parol_runtime` with default features from `build-dependencies` are conflicted.

closed 2025-03-05 by dalance

dalance, 2025-03-05:
@jsinger67 

How do you think about this issue?
I already resolved it by adding `build` feature to disable parol of `build-dependencies`,
but fundamental solution may be adding feature for parol_runtime to parol like below:

```Cargo.toml
[dependencies]
parol_runtime = {version = "2.2", features = ["regex_automata"], default-features = false}

[build-dependencies]
# pass regex_automata to parol_runtime
parol = {version = "3.0", features = ["regex_automata"], default-features = false}
```

jsinger67, 2025-03-05:
Parole itself does not use the regex_automata feature for its own reference of parol_runtime.
That's why parole v3 re-exports parol_runtime to avoid the problem you describe because you only need to use parol as build-dependency and thus you can't create a conflicting combination of features.
Nevertheless, I was not aware that both dependencies and build-dependencies can be active at the same time.
If this is the case you will always end up with cargo errors.
On the other hand, providing an adequate feature for parol causes much higher test efforts.
I will have to investigate the topic further.

dalance, 2025-03-05:
@jsinger67 

OK. In this case, user of `veryl-parser` don't need to execute parol build. So I think the current solution is sufficient.
Maybe it is more useful that this workaround is described in the parol book.
(I also didn't know both dependencies become to be active, so felt very strange behavior)

jsinger67, 2025-03-05:
Yes, I agree. I will document this behavior thoroughly in the book 😊