veryl-lang/veryl #1353: No error is reported when refering a generic parameter via scoped identifier

taichi-ishitani, 2025-03-07:
Reference to a generic parameter via a scoped identifier should cause an error but not.

[image]

taichi-ishitani, 2025-03-07:
Reference to a local variable in a function also causes no error.

[image]

taichi-ishitani, 2025-03-07:
Reference to an indentifier in a module also causes no error.

[image]

taichi-ishitani, 2025-03-07:
Reference to struct member via the struct type.

[image]

[image]

taichi-ishitani, 2025-03-07:
Reference to a variable in a inteface.

[image]

closed 2025-03-08 by dalance

nananapo, 2025-04-20:
Is it intended to prohibit reference to a variable in submodule?

[image]

nananapo, 2025-04-20:
By the way, Playground(nightly) can build this code.
My local environment can't...
[image]

taichi-ishitani, 2025-04-20:
`u` is a module instance so behavior of nightly build is invalid.

taichi-ishitani, 2025-04-20:
I add the unit test below and I confirm that the `InvisibleIndentifier` error is reported expectedly.
This behavior is intentional.

```rust
    let code = r#"
    module ModuleA {
        let _a: logic = 0;
    }
    module ModuleB {
        inst u: ModuleA;
        let _b: logic = u._a;
    }
    "#;

    let errors = analyze(code);
    assert!(matches!(
        errors[0],
        AnalyzerError::InvisibleIndentifier { .. }
    ));
```