veryl-lang/veryl #1364: Change dependency syntax

dalance, 2025-03-08:
The current dependency syntax has some problems.

* Imported package name is implicit
* Can't handle multiple packages in a repository

So I think the following syntax as the same as `Cargo.toml` of Rust is better.

```toml
[dependencies]
# basic syntax
veryl_sample = {version = "0.1.0", git = "https://github.com/veryl-lang/veryl_sample"}

# rename importing name
veryl_sample2 = {version = "2.0.0", git = "https://github.com/veryl-lang/veryl_sample", package = "veryl_sample"}

# multiple packages in a repository
veryl_prj1 = {version = "0.1.0", git = "https://github.com/veryl-lang/veryl_sample"}
veryl_prj2 = {version = "0.1.0", git = "https://github.com/veryl-lang/veryl_sample"}
veryl_prj3 = {version = "0.1.0", git = "https://github.com/veryl-lang/veryl_sample"}

# override by local path
veryl_sample3 = {version = "0.1.0", git = "https://github.com/veryl-lang/veryl_sample", path = "../../veryl_sample"}

# path only ( can't be published )
veryl_sample4 = {path = "../../veryl_sample"}
```

taichi-ishitani, 2025-03-08:
How about adding `github` field?
This is a syntax suggar of `git` field:

```toml
# same with git = "https://github.com/veryl-lang/veryl_sample"
veryl_prj1 = {version = "0.1.0", github = "veryl-lang/veryl_sample"}
```

dalance, 2025-03-10:
I adjusted some syntax.

```toml
[dependencies]
# basic syntax
veryl_sample = {github = "veryl-lang/veryl_sample", version = "0.1.0"}

# rename importing name
veryl_sample2 = {github = "veryl-lang/veryl_sample", project = "veryl_sample", version = "0.2.0"}

# multiple packages in a repository
inner_prj1 = {github = "veryl-lang/veryl_sample", version = "0.1.0"}
inner_prj2 = {github = "veryl-lang/veryl_sample", version = "0.1.0"}
inner_prj3 = {github = "veryl-lang/veryl_sample", version = "0.1.0"}

# override by local path
veryl_sample3 = {github = "veryl-lang/veryl_sample", version = "0.1.0", path = "../../veryl_sample"}

# path only ( can't be published )
veryl_sample4 = {path = "../../veryl_sample"}

# arbitrary git repository
veryl_sample5 = {git = "https://example.com/veryl_sample", version = "0.1.0"}
```

PR #1373 (2025-03-24): Change dependency syntax
Closes #1364
closed 2025-03-24 by dalance