veryl-lang/veryl #1370: [Feature] Allow to use non generic module/interface/package as proto boundary

taichi-ishitani, 2025-03-09:
For non generic module/interface/package, there are no meanings to define a prototype for that.
I think it would be nice to use such components as a proto boundary.
For #793, there are cases that non generic interface instances are given to generic code block.

dalance, 2025-04-14:
I can't understand this necessity yet.
For example, the following `ModuleB` can take only `ModuleA` as a generic parameter.
So is it the same as `module ModuleB { inst u: ModuleA }`?

```systemverilog
module ModuleA {}
module ModuleB::<T: ModuleA> {
    inst u: T;
}
module ModuleC {
    inst u: ModuleB::<ModuleA>;
}
```

taichi-ishitani, 2025-04-14:
> I can't understand this necessity yet.

This feature is for `generics for code block` #793.
Interface specified for generic parameter may has no generic parameters.
Therefore, non generic interface needs also to be used as generic boundary.

taichi-ishitani, 2025-04-15:
> inst X

As my understanding, `inst` generic boundary also requires a prototype.
Therefore, this feature is needed to use non generic interface as a type of `inst` generic parameter.

dalance, 2025-04-15:
The combination becomes like below, and only 5 patterns are meaningful.
So I think it is not necessary to prioritize symmetricity.

| |module|interface|package|
|-|-|-|-|
|generic| o | o | o |
|non generic| | | |
|inst generic| |o| |
|inst non generic| |o| |

taichi-ishitani, 2025-04-15:
Ok, I understand. I will remove non meaningful patterns.

taichi-ishitani, 2025-04-15:
Use of non generic interface as generic inst parameter has already been supported.
https://github.com/veryl-lang/veryl/blob/4eced5e2e98039af3166aeb218f1c9a35833b523/crates/analyzer/src/handlers/check_type.rs#L92

We only have to implement check for non meaningful generic parameters.

closed 2025-04-16 by dalance